arendst/Tasmota #3470: HTTP routing

Dees7, 2018-08-13:
HTTP interface unavailable through routing.

I can connect to ESP from local-net. 

Also I have vpn-server on my router.There is routing used to connect between networks.  In case I connect to vpn 

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : 
Sonoff Basic, Sonoff Dual R2, NodeMCU
- [x] Provide the output of command ``status 0`` :
```
08:25:14 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["MCU"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:25:14 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:11:04","StartupUTC":"2018-08-13T03:14:10","Sleep":0,"BootCount":131,"SaveCount":576,"SaveAddress":"F4000"}}
08:25:14 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.1.1c","BuildDateTime":"2018-08-05T21:35:53","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
08:25:14 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"172.16.95.130","LogPort":514,"SSId":["195","95"],"TelePeriod":60,"SetOption":["00008009","99958000","00000001"]}}
08:25:14 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":466,"Free":536,"Heap":11,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0F8827B4","00000000","25021796","00000000"]}}
08:25:14 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"nodemcu","IPAddress":"172.16.95.121","Gateway":"172.16.95.1","Subnetmask":"255.255.255.0","DNSServer":"<mine1>","Mac":"EC:FA:BC:11:00:00","Webserver":2,"WifiConfig":1}}
08:25:14 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.95.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07D80D","MqttUser":"wiot","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:25:14 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 13 03:25:14 2018","Local":"Mon Aug 13 08:25:14 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":5,"Sunrise":"09:40","Sunset":"00:10"}}
08:25:14 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-08-13T08:25:14","Switch1":"1","DS18B20":{"Temperature":40.81},"BMP280":{"Temperature":26.08,"Pressure":984.08},"ENERGY":{"Total":55.824,"ActivePower":32.68,"ApparentPower":35.31,"ReactivePower":-12.65,"Frequency":50.00,"Factor":0.92,"Voltage":234.07,"Current":0.151,"Phase angle":-21.72,"Import Active Power":55.82,"Export Active Power":0.00,"Import Reactive Power":0.08,"Export Reactive Power":49.85,"Total Reactive Power":49.93},"TempUnit":"C"}}
08:25:14 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-08-13T08:25:14","Uptime":"0T00:11:04","Vcc":3.075,"POWER":"0","Wifi":{"AP":2,"SSId":"95","RSSI":60,"APMac":"60:31:97:11:00:00"}}}:"172.16.95.1","Subnetmask":"255.255.255.0","DNSServer":"89.237.19.25","Mac":"EC:FA:BC:11:00:00","Webserver":2,"WifiConfig":1}}
08:25:14 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.95.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07D80D","MqttUser":"wiot","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:25:14 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 13 03:25:14 2018","Local":"Mon Aug 13 08:25:14 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":5,"Sunrise":"09:40","Sunset":"00:10"}}
08:25:14 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-08-13T08:25:14","Switch1":"1","DS18B20":{"Temperature":40.81},"BMP280":{"Temperature":26.08,"Pressure":984.08},"ENERGY":{"Total":55.824,"ActivePower":32.68,"ApparentPower":35.31,"ReactivePower":-12.65,"Frequency":50.00,"Factor":0.92,"Voltage":234.07,"Current":0.151,"Phase angle":-21.72,"Import Active Power":55.82,"Export Active Power":0.00,"Import Reactive Power":0.08,"Export Reactive Power":49.85,"Total Reactive Power":49.93},"TempUnit":"C"}}
08:25:14 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-08-13T08:25:14","Uptime":"0T00:11:04","Vcc":3.075,"POWER":"0","Wifi":{"AP":2,"SSId":"95","RSSI":60,"APMac":"60:31:97:11:00:00"}}}
```

**To Reproduce**
0. Start OpenVPN server on Router
1. Connect ESP to WiFi
2. Connect Client from internet to openVPN server
3. Try to open web-interface.

**Expected behavior**
have Web-interface shown

**Screenshots**

[image]

**Additional context**
Ping works well.

Frogmore42, 2018-08-13:
Before clicking the search box in the template, it is a good idea to actually do a search  😀

Core 2.4.1 is known to have issues. Core 2.4.0 is known to be bad.  Core 2.3.0 is the current stable one and appears to even work over VPN.

https://github.com/arendst/Sonoff-Tasmota/issues/2663

Jason2866, 2018-08-13:
Maybe with core 2.4.2?

Frogmore42, 2018-08-13:
2.4.2 is supposed to be better, but not many people have tried it.  PlatformIO just made an update, but not a release, that makes it possible to use 2.4.2.  I have not had a chance to try it yet.

Dees7, 2018-08-13:
2.4.2 doesn't wok. 2.3.0 works well. Thanks.

Jason2866, 2018-08-13:
I am testing 2.4.2 (with VSC Platformio) on some devices since yesterday.
Looks good so far, but i doesnt use it via VPN. 
I control my devices via a HA System. This System i can reach over a secure way...

ascillato2, 2018-08-18:
Closing this issue as Dees7 said that with 2.3.0 it worked. Thanks

closed 2018-08-18 by ascillato2