arendst/Tasmota #3471: Timer (ARM unchecked)

chris8837, 2018-08-13:
Maybe it was a bad flash but I have a timer setup for a sonoff s20 and Every so often it won’t turn off or on the device. 

When I go into the timer area I noticed that ARM is u checked for both on and off function. This has happened 3 times now so I see a pattern here. 

I def checked the arm and save them a reset just in case. But for some reason it randomly unchecked itself and the light I’m using it on doesn’t turn off.

Jason2866, 2018-08-13:
Without log files and further exact information nobody can help.

chris8837, 2018-08-13:
Yeah sorry about that. I did some trouble shooting and I think I’ve got it?

So I’m using HASS and I deleted the retain: true completely from all the switches. I’ve read that when home assistant tries to retain it messes up things?

I completely removed it and set powerretain on tasmota to powerretain 1. 

Does this sound right?  Seems to be working.

Jason2866, 2018-08-13:
Glad to hear you solved it. I am pretty sure the trouble you had was the retained messages

curzon01, 2018-08-14:
**(Please, remember to close the issue when the problem has been addressed)**

closed 2018-08-14 by chris8837