arendst/Tasmota #3474: Sonoff bridge v2 connect failed to mosquitto on RasPi wheezy

krifi07, 2018-08-13:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X ] Hardware used : 
RasPi running wheezy, Sonoff RF bridge V2
- [X] Provide the output of command ``status 0`` :
```
10:46:49 CMD: status 0
10:46:49 RSL: stat/sonoff1/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff1"],"Topic":"sonoff1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:46:49 RSL: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:47","StartupUTC":"2018-08-13T09:39:02","Sleep":0,"BootCount":28,"SaveCount":40,"SaveAddress":"F4000"}}
10:46:50 RSL: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-08-11T16:23:07","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
10:46:50 RSL: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Christian",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
10:46:50 RSL: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":559,"Free":444,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
10:46:50 RSL: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff1-2103","IPAddress":"192.168.0.8","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:CA:C8:37","Webserver":2,"WifiConfig":3}}
10:46:50 RSL: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.17","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CAC837","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:46:50 RSL: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 13 09:46:50 2018","Local":"Mon Aug 13 10:46:50 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:40","Sunset":"20:10"}}
10:46:50 RSL: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2018-08-13T10:46:50"}}
10:46:50 RSL: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2018-08-13T10:46:50","Uptime":"0T00:07:48","Vcc":3.438,"Wifi":{"AP":1,"SSId":"Christian","RSSI":58,"APMac":"BC:05:43:9D:F6:91"}}}
10:46:54 MQT: Attempting connection...
10:47:09 MQT: Connect failed to 192.168.0.17:1883, rc -4. Retry in 10 sec
```

Hello everybody,
thanks for providing all the efforts to develop Tasmota and support this project.

I got an RF bridge V2, flashing was running fine, and now I wanted to connect it to my FHEM HA system running on a RasPi. When checking the mosquitto server on the RasPi with MQTT.fx or an android client messages are fine and the mqtt server seems to be running correctly.
However, the Sonoff bridge doesn't connect to the server, see above messages.

I'm sitting over this now for 1 1/2 days, therefore now dare to ask everybody.

Thanks in advance
Christian

**(Please, remember to close the issue when the problem has been addressed)**

Dees7, 2018-08-13:
Can you ping 192.168.0.8  (Sonoff) from 192.168.0.17 (Rasp)?

krifi07, 2018-08-13:
Yes, no problem.

Thanks!

Dees7, 2018-08-13:
Can you install tcpdump onto Rasp and start 
sudo tcpdump -n  -i $(ip route | awk '/192.168\.0\.0/{print $3}')  'host 192.168.0.8 and port 1883'

also try turn on logging mosquito

/etc/mosquitto/mosquitto.conf
...
log_type all
log_dest file /var/log/mosquitto/mosquitto.log
...

sudo mkdir /var/log/mosquitto 
sudo chown mosquitto.  /var/log/mosquitto

/etc/init.d/mosquitto stop
wait 5 sec (masquito can't start cause port 1883 used)
/etc/init.d/mosquitto start

if both logs exist provide them.

Also check  and provide
sudo iptables  -L

arendst, 2018-08-13:
See https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview#troubleshooting regarding rc -4

krifi07, 2018-08-13:
Thanks.
I already have seen that, however cannot figure out what I should do as a consequence.
When I changed the IP address of the mosquitto server the error code changed to rc -2, so to me this means that something with mosquitto seems to be wrong, however, as said, two other clients connect successfully, also the test with two parallel ssh connections to the Raspi (one listening, one publishing) were fine.

BR
Christian

krifi07, 2018-08-13:
Hi,

have a look to this, that sounds quite promising to me:
https://github.com/knolleary/pubsubclient/issues/159
But actually I don't want start compiling on my own...never went that far in Linux....
What do you think?

Frogmore42, 2018-08-13:
@krifi07 you are using core 2.4.1 which is known to have many problems with networking.  Switch to 2.3.0 and see if that fixes it.  There is info in the wiki and many other issues about core versions and how to use different ones.

krifi07, 2018-08-13:
@Frogmore42 thanks, reflashed, still same error.
I simply don't want to shoot my entire FHEM setup which is working since years perfectly alright by upgrading to Stretch.

Frogmore42, 2018-08-13:
The core number refers to the esp8266 Arduino core.  What did you flash it with and where is the status 0.

krifi07, 2018-08-13:
yes, that's clear, I flashed it to the Sonoff bridge,
Status 0:
19:04:09 RSL: stat/sonoff1/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff1"],"Topic":"sonoff1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:04:09 RSL: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:18:03","StartupUTC":"2018-08-13T17:46:06","Sleep":0,"BootCount":36,"SaveCount":48,"SaveAddress":"F4000"}}
19:04:09 RSL: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-08-13T18:20:54","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:04:09 RSL: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Christian",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
19:04:09 RSL: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","23B6179E","00000000"]}}
19:04:09 RSL: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff1-2103","IPAddress":"192.168.0.8","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:CA:C8:37","Webserver":2,"WifiConfig":3}}
19:04:09 RSL: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.17","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CAC837","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:04:09 RSL: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 13 18:04:09 2018","Local":"Mon Aug 13 19:04:09 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:40","Sunset":"20:10"}}
19:04:09 RSL: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2018-08-13T19:04:09"}}
19:04:09 RSL: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2018-08-13T19:04:09","Uptime":"0T00:18:03","Vcc":3.153,"Wifi":{"AP":1,"SSId":"Christian","RSSI":52,"APMac":"BC:05:43:9D:F6:91"}}}

krifi07, 2018-08-13:
and here's the mosquitto logging:

1534180654 mosquitto version 0.15 (build date 2018-04-01 162621+0000) starting
1534180654 Opening ipv4 listen socket on port 1883.
1534180654 Opening ipv6 listen socket on port 1883.
1534180654 Warning Address family not supported by protocol
1534180656 New connection from 192.168.0.8.
1534180656 Invalid protocol MQTT in CONNECT from 192.168.0.8.
1534180656 Socket read error on client (null), disconnecting.
1534180683 New connection from 192.168.0.8.
1534180683 Invalid protocol MQTT in CONNECT from 192.168.0.8.
1534180683 Socket read error on client (null), disconnecting.
1534180705 New connection from 192.168.0.17.
1534180705 New client connected from 192.168.0.17 as NetMQTTpm12158.
1534180710 New connection from 192.168.0.8.
1534180710 Invalid protocol MQTT in CONNECT from 192.168.0.8.
1534180710 Socket read error on client (null), disconnecting.
1534180737 New connection from 192.168.0.8.
1534180737 Invalid protocol MQTT in CONNECT from 192.168.0.8.
1534180737 Socket read error on client (null), disconnecting.
1534180765 New connection from 192.168.0.8.
1534180765 Invalid protocol MQTT in CONNECT from 192.168.0.8.
1534180765 Socket read error on client (null), disconnecting.

When stopping mosquitto, the error code changes from rc -4 to rc -2.

sudo iptables -L liefert nur:

pi@raspberrypi ~ $ sudo iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

Still need the tcpdump?

Frogmore42, 2018-08-13:
@krifi07 it seems that you will need to compile the Tasmota code.  The default MQTT code appears to not be compatible with your MQTT server.

See here for the choices for MQTT:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_01_mqtt.ino

The alternative would be to update your MQTT server.

ascillato2, 2018-08-24:
Hi,

@krifi07 

Have you managed to solve your issue?

krifi07, 2018-08-24:
Hi,

not yet. Instead of using my existing Intertechno sockets, I will get a Sonoff switch and try that one to use with Tasmota first.
If that doesn't work, my next step would have been to use ESPeasy.

Actually I'm wondering that it seems I'm quite lonely with this problem.
I would have guessed that there should still be many Wheezy users outside.

Thanks.

Frogmore42, 2018-08-24:
Wheezy is quite old now it seems, which is probably why no one else is seeing this issue.  I don't believe using a Sonoff device will change your results.  The logs indicate you have incompatible MQTT protocols.  This is a software issue.  You can fix it by upgrading your old broker.  This might help:
https://raspberrypi.stackexchange.com/questions/80051/how-to-upgrade-mosquitto-mqtt-to-the-latest-version

Or, you might be able to select a different MQTT client in the Tasmota that knows how to speak the older format that your broker knows how to understand.

krifi07, 2018-08-24:
Thanks a lot again.
Let me check, however it will take a while as I'm on holiday now.
And the project's priority got lower again, as I wanted to control my garden irrigation with it.

I will let you know if it helped.

If not I might use some time in autumn to upgrade my Pi.

ascillato, 2018-08-28:
Hi,

@krifi07

Have you managed to solve your issue?

krifi07, 2018-09-02:
I tried upgrading mosquitto to the latest version according to the description above [https://raspberrypi.stackexchange.com/questions/80051/how-to-upgrade-mosquitto-mqtt-to-the-latest-version](url)

however the wheezy list is no longer maintained on the rep, therefore not useful for me.

This again shows that the most reasonable step is to upgrade my system from wheezy to stretch.

I think we can close the topic here. Thanks a lot for all your help!

closed 2018-09-03 by ascillato2