arendst/Tasmota #3475: Start time / stop time on each timer

mcc05, 2018-08-13:
I'm using tasmota as a thermostat, got that working ok with rules.

The next problem is timers, so I have the following:
rule2
on Clock#Timer=1 do mem 1 endon on Clock#Timer=2 do mem 0 endon
on Clock#Timer=3 do mem 1 endon on Clock#Timer=4 do mem 0 endon
on Clock#Timer=5 do mem 1 endon on Clock#Timer=6 do mem 0 endon
on Clock#Timer=7 do mem 1 endon on Clock#Timer=8 do mem 0 endon
on Clock#Timer=9 do mem 1 endon on Clock#Timer=10 do mem 0 endon
on Clock#Timer=11 do mem 1 endon on Clock#Timer=12 do mem 0 endon
on Clock#Timer=13 do mem 1 endon on Clock#Timer=14 do mem 0 endon

which does equate to controlling 1 on and off time period per day of the week, however most thermostats at least allow for 3 schedules per day, morning / afternoon / nighttime, but with only 16 timers this is not possible it would need 21.

Would it be possible to have a start and stop time option on each timer?

Jason2866, 2018-08-13:
Just my 2 cents. Complex(er) things should be done in a HA system.

tbagro, 2018-08-14:
if short intervals can be done
rule1 on clock#timer=1 do backlog power1 %value%; ruletimer1 600 endon on rules#timer=1 do power1 off endon

ascillato2, 2018-08-21:
Hi,

If you think that your question has been answered, please close this issue. Thanks.

closed 2018-08-22 by ascillato2

ascillato, 2018-08-22:
Hi,

Sorry. I close the issue as you did not answered before and you have received 2 valid and different ways of solving your request.

Timers are triggers, not conditionals in Tasmota.
If you need conditionals, you can use rules and the variable %time%. 
There are some examples of that in issues and in the wiki.

ascillato, 2018-08-22:
If you believe this issue needs to be reopen, just ask.

mcc05, 2018-08-22:
Its fine would just be a nice feature to finish off a full working thermostat using the timers for start/stop periods over rules. I am looking at porting the Martin Harizanov. thermostat version over to sonoff which I think is probably a better fit than Tasmota for a specific function like I need.

chivas03, 2022-12-07:
> Its fine would just be a nice feature to finish off a full working thermostat using the timers for start/stop periods over rules. I am looking at porting the Martin Harizanov. thermostat version over to sonoff which I think is probably a better fit than Tasmota for a specific function like I need.

Hello,
have you finished your project regarding thermostat, rules and timers?
Would be fine if you can share your result as I want to start similar project :-)

Regards