arendst/Tasmota #3477: Sonoff 4ch Pro R2 not working with 2 sensors

fanofsonoff, 2018-08-13:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X] Hardware used : Sonoff 4Ch Pro R2
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
23:32:32 RSL: Group 0, Index 1, Command STATUS, Data 0
23:32:32 MQT: stat/garage/STATUS = {"Status":{"Module":23,"FriendlyName":["Garage1","Garage2","Garage3","Garage4"],"Topic":"garage","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
23:32:32 MQT: stat/garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.1.1/sonoff-allsensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:11","StartupUTC":"2018-08-13T22:32:21","Sleep":0,"BootCount":75,"SaveCount":277,"SaveAddress":"F6000"}}
23:32:32 MQT: stat/garage/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-31T17:59:09","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:32:32 MQT: stat/garage/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXX",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
23:32:32 MQT: stat/garage/STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","03B6179E","00000000"]}}
23:32:32 MQT: stat/garage/STATUS5 = {"StatusNET":{"Hostname":"garage-XXXX","IPAddress":"192.168.1.XXX","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":3}}
23:32:32 MQT: stat/garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.XXX","MqttPort":1883,"MqttClientMask":"XXXXXX","MqttClient":"XXXXXX","MqttUser":"XXXX","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:32:32 MQT: stat/garage/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 13 22:32:32 2018","Local":"Mon Aug 13 23:32:32 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:40","Sunset":"20:10"}}
23:32:32 MQT: stat/garage/STATUS10 = {"StatusSNS":{"Time":"2018-08-13T23:32:32","Switch1":"ON","Switch2":"ON"}}
23:32:32 MQT: stat/garage/STATUS11 = {"StatusSTS":{"Time":"2018-08-13T23:32:32","Uptime":"0T00:00:11","Vcc":3.182,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXX","RSSI":94,"APMac":"XX:XX:XX:XX:XX:XX"}}}
23:32:35 MQT: tele/garage/STATE = {"Time":"2018-08-13T23:32:35","Uptime":"0T00:00:14","Vcc":3.182,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXX","RSSI":90,"APMac":"XX:XX:XX:XX:XX:XX"}}
23:32:35 MQT: tele/garage/SENSOR = {"Time":"2018-08-13T23:32:35","Switch1":"ON","Switch2":"ON"}
**(Please, remember to close the issue when the problem has been addressed)**

I am trying to use Sonoff 4Ch Pro R2 as Garage Door Opener(2 car garage with 2 separate doors). I turned first 2 buttons to be in inching mode. It is all working good.

Now, I  need to add 2 reed switch sensors to the device to detect if the doors are open or not.

I turned serial logging to 0(zero).

I am trying to use RX and TX as switches for the 2 sensors I need.

RX is working flawlessly. If I add another sensor and connect to TX, when the device boots, it lights the first 3 orange lights and does not connect to my router. The buttons dont work. Basically it seems stuck in booting.

However, if I turn-on the device and then connect the sensors after it is on, both the RX and TX work flawlessly.

I saw the same behavior with SDA(GPIO2).

I made sure to have the right SwitchTopic and SwitchMode, I know that is not the issue because everything works if I connect reed switches after Sonoff has booted.

Please help.

Frogmore42, 2018-08-13:
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#gpio-overview

TX is an output during boot and there is nothing you can do to prevent it.  If you have it shorted to ground (ie you reed switch is closed) it will prevent the esp8266 from booting.

The esp8266 does not have lots of pins that are available for general purpose GPIO.  If you need more than about 5 GPIO, you will have to be extra careful and really should consider using multiple esp8266 devices for that scenario.

There are other MCU's that have many GPIOs, but none that are as cheap as the esp8266 and have Wi-Fi built in.  SDA should be fine though.

fanofsonoff, 2018-08-14:
If it makes any difference, Sonoff 4Ch Pro R2 is ESP8285.

All I need is 2 GPIO. Serial Logging is turned off. RX works. Both TX and SDA does not work. Any other GPIO I could use?

Frogmore42, 2018-08-14:
https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html

With a 4 channel device, you are out of pins that are easy to use.  SDA (GPIO 2) has to be held high to boot properly, so it is not a good choice for a switch that might be closed when power comes on.

You might be able to use one of the GPIOs that are normally used for flash, since you have an esp8265.  I don't know how easy that will be to find.  But the information is available with a simple search.  The template won't allow you to use those pins, so you will likely need to change it and recompile.  As I have said before, the esp8266 is not a good choice for projects that need a lot of pins.

Jason2866, 2018-08-14:
You could use GPIO 16. You have to solder direct on the ESP8285 -> no fun job.

fanofsonoff, 2018-08-14:
Thanks Frogmore42 and Jason2866 for your comments.

closed 2018-08-14 by fanofsonoff