arendst/Tasmota #3480: rainbow with ws2812 leds

meesvandenberg, 2018-08-14:
Hi,

I have managed to get my sonoff s20 working with my led strip, but i'm not satisfied with the schemes and functionality. Specifically the rainbow on scheme 11. Is there a way to implement my own strings of code for more functionality?

I thought maybe i could edit the firmware.bin file but i am not a programmer so i didn't succeed. I saw tasmota was build with the neopixelbus library so maybe I can build further on that. Anyone who can help me?

roguestreak, 2018-08-14:
Probably not exactly what you are after, but you can set each pixel individually in one hit using the Backlog command, eg.
`Backlog Led1 FF0000; Led2 FF00FF; Led3 00FF00 etc...`

If you have a home automation system with Node Red you could dynamically create and deploy the arrays that way, some discussion here for my solar pv / home consumption usage meter here:
https://groups.google.com/forum/#!topic/sonoffusers/hqwm-5yaa9c/discussion
(code tweaked a little since then)

meesvandenberg, 2018-08-14:
@roguestreak Thanks for your response!! That's a possible option but I have 178 Leds and I don't want to program them all individually.

mike2nl, 2018-08-14:
@meesvandenberg 
When you not a programmer it isn't an easy task.

What you can do is make a copy of the original file and change that one.
In your case here: -> xplg_ws2812.ino
Make a copy from the file sdrv_04_light.ino too if you change anything in it.

If you use Arduino IDE verify it, compile and upload it and test your changings. 

But when you killed somethings i think you wont get any support for it. 
The only way is then copy back the copied/backuped files and verify it, compile and upload it again.

So now let's do crazy ;-):
1. You can look into sdrv_04_light.ino (Line: 36..50, 54)
2. Then look into xplg_ws2812.ino (Line: 66..72), there are the 
    additional 7 schemes defined in sdrv_04_light.ino (Line: 54).

There you have color defintions and the count.
Example in Line 71: 
-> WsColor kRainbow[7] = { 255,0,0, 255,128,0, 255,255,0, 0,255,0, 0,0,255, 128,0,255, 255,0,255 };
You see the color definiton in RGB or GRB decimal values? 
-> G,R,B next one G,R,B ..... {255,0,0, 255,128,0, ..... };
You count 7 of them? 
-> kRainbow[7] .... yes there are seven (7) of them

These seven (7) RGB defines can be changed. Lets say the first one is:
255,0,0, and change it t0 255,0,128. You got the idea? 

You know the color defines:
R =Red, can be 0...255
G =Green, can be 0...255
B = Blue, can be 0...255

Help for color defines by the basics, look here:
-> https://www.w3schools.com/colors/colors_rgb.asp

-> For the different types of ws2812 LED's look into xplg_ws2812.ino (Line: 27..55)

Different types of RGB order can be: 
-> NEO_RGB, NEO_GRB, NEO_BRG, NEO_RBG, NEO_RGBW, NEO_GRBW.

Default in the tasmota user_config.h is: 
-> NEO_GRB

What you have used in your user_config.h (the next two lines are the original one)?
#define USE_WS2812
  #define USE_WS2812_CTYPE     NEO_GRB

Here/there (xplg_ws2812.ino) you can add or change things in a more 
easy way without any deep coding experience.

!!! Think about Line 54 in sdrv_04_light.ino: #define WS2812_SCHEMES 7 
When you will 'add' a scheme you must increase that number from 7 to 8 as an example.

But first: change only one scheme for a first impression and testing purpose.

> When you are not a programmer and it looks creapy for you ;-) 
 ...DON'T touch it.
Learn, read and read again first and then think twice what you are doing.
-> Think about that twice, please. A serious and friendly meaned warning. <-

roguestreak, 2018-08-15:
>  I have 178 Leds and I don't want to program them all individually.
If you find the right code you don't have to do them individually, the function can do it for you.  Just found this one, which goes VERY heavily into depth, but the final example could be readily tweaked to create a dynamically-generated array of rainbow colours to smoothly fit however many LEDs you have
https://krazydad.com/tutorials/makecolors.php
I think it could be something along the lines of this
```
var pixelArr = new Array("null")

function RGB2Color(r,g,b)
{
  return byte2Hex(r) + byte2Hex(g) + byte2Hex(b);
}

function byte2Hex(n)
{
    var nybHexString = "0123456789ABCDEF";
    return String(nybHexString.substr((n >> 4) & 0x0F,1)) + nybHexString.substr(n & 0x0F,1);
}

function createColourArray(numPixels,phase)
{
    if (phase == undefined) phase = 0;
  center = 128;
  width = 127;
  frequency = Math.PI*2/(numPixels+1);
  for (var i = 1; i <= numPixels; ++i)
  {
	pixelArr[i] = "000000;"; //pre-fill blank
	red   = Math.sin(frequency*i+2+phase) * width + center;
	green = Math.sin(frequency*i+0+phase) * width + center;
	blue  = Math.sin(frequency*i+4+phase) * width + center;
	pixelArr[i] = RGB2Color(red,green,blue);
  }
}
```
If you call "createColourArray(178)", that SHOULD(!) (might?) give you a dynamically generated smooth array of rainbow goodness, one element per LED.  Next step, which I haven't nutted out, is to loop through that array in such a way that you grab 30 elements out and feed them to the Backlog, then grab the next 30, etc.

ascillato, 2018-08-23:
Similar issues added for reference:

#285
#2906
#39
#3480

more effects at: https://github.com/kitesurfer1404/WS2812FX

encor50, 2018-09-10:
how do i call the rainbow effect with home assistant or other effects ? thanks

ascillato, 2018-09-10:
Please, see examples in the wiki

ascillato2, 2018-12-03:
@dimalo feature request #4512 added to this list of ws2812 requests:

**Have you look for this feature in other issues and in the wiki?**
Yes and didn't find something.

**Is your feature request related to a problem? Please describe.**
I am enjoying Tasmota quite a while now with a lot of Sonoffs - THANK YOU.
I have some Sonoff SVs controlling WS2812-like LED strips.
I would like to have a fading effect when turning it on. 

**Describe the solution you'd like**
Instead of fading all LEDs at once, it would look very nice if at first just one lid up, then 2 then 3... Depending on the fading speed the whole strip would come to life from left to right. Or from right to left. Or even giving it a starting point and then it shall fade to the left and right from the starting point.

In other words: The fading effect should not be accomplished by controlling the brightness of all LEDs, but the number of lid LEDs. 
This can also be enhanced with the number and the brightness of each LED.

I imagine three new options:
1. SeqFade: boolean to activate the fading method
2. SeqFadeMode: 0=left to right, 1=right to left, 2=both directions, 3=both directions inverted
3. SeqFadeStart (Only for SeqFadeMode=2): Number, which is < Pixel count, defining LED to start the fade from - if Pixel count % 2 == 0 then first animation frame lights up SeqFadeStart and SeqFadeStart + 1 (see example below)

**Describe alternatives you've considered**
The alternative would be to send the strip a lot of PIXELS commands, which doesn't give a nice animation and doesn't give the option of fading direction.

**Additional context**
Here is an example showing the state of a 10 LED strip in each animation frame:
```
x -> ON
- -> OFF

Left to Right:
----------
x---------
xx--------
xxx-------
xxxx------
xxxxx-----
xxxxxx----
xxxxxxx---
xxxxxxxx--
xxxxxxxxx-
xxxxxxxxxx

Right to Left:
----------
---------x
--------xx
-------xxx
------xxxx
-----xxxxx
----xxxxxx
---xxxxxxx
--xxxxxxxx
-xxxxxxxxx
xxxxxxxxxx

Both Directions with SeqFadeStart == 5:
----xx----
---xxxx---
--xxxxxx--
-xxxxxxxx-
xxxxxxxxxx

Both Directions inverted with SeqFadeStart == 5:
x--------x
xx------xx
xxx----xxx
xxxx--xxxx
xxxxxxxxxx
```
SeqFadeMode 2 and 3 should be executed either with half speed or with fading in brightness of each LED in 2 steps (0 -> 50% -> 100%)

I WOULD LOVE THAT FEATURE!

ascillato2, 2018-12-03:
@pinoyyid feature request #4447 added to this list of ws2812 requests:

**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
I have a 65 pixel ws2812 linked to a nodeMCU as a headboard. Logic in my controller divides this into 5 banks of 13 pixels. If I want banks 1,2 & 3 in blue, and banks 4 and 5 in red. I send 5 backlog commands...
```
backlog led1 0000ff; led2 0000ff ... led13 0000ff
backlog led14 0000ff; led15 0000ff ... led26 0000ff
...
backlog led52 ff0000; led53 ff0000 ... led65 ff0000
```

**Describe the solution you'd like**
Could the `LEDx rrggbb` command support a range, eg. `LED1,3,5,7-10 rrggbb`. This would simplify my multiple backlogs into a two commands, `LED1-39 0000ff ; LED40-65 ff0000`

ascillato2, 2018-12-14:
@jascdk feature request #4622 added to this list of ws2812 requests:

Hi.

Would it be possible to integrate a "Breathable LED effect" in the LED light options.

Mainly for use in projects to indicate that MQTT / wifi etc. is not connected.

I used this code in another project linked below:

https://arduining.com/2015/08/20/nodemcu-breathing-led-with-arduino-ide/

It gives a nice effect - a bit more cool than just standard blinking :)

ozett, 2018-12-24:
maybe inspiration for effects from here: 

https://github.com/kitesurfer1404/WS2812FX

?

spitzbub250, 2019-03-11:
Is there a way to let only one led fade? I tried to set 1 - 3 leds from off to a specified color, but there is no fade effect. The Fade effect only is shown, if i put all pixels on. 

Is there a way to do this?

ascillato2, 2019-03-14:
From @computerkoenig at #5470:

It would be really nice to have a candlelight effect for normal RGB LEDs.
For example now I have a esp8266 RGB Strip controller and only a "white" LED connected.
It would be really nice to have a candlelight for this white led. Like a little bit flickering.

Do you need more informations?

ozett, 2019-03-21:
there are cumulated fetaure-Requests,
hard to get the actual level of coding.

@ascillato/ascillato2:
Is there something to use now? where can i check this? is your request still only a request?

ascillato, 2019-03-21:
@ozett

Sorry, I don't understand your question. This issue is a feature request for more effects, done by other users. Not me. 

The features supported at this moment by Tasmota are explained in the wiki. See commands.

M0ebiu5, 2019-04-23:
Maybe you could also consider the following mode:

In my kitchen i will use led pixel strips to illuminate the work area (L shape, 2 x 3m). And i've integrated 2 VL53L1X sensors in the bottom area of the kitchen.

If someone approaches the work area, the strip should fade in like SeqFadeMode 2, but the active leds should only be apr. 1m wide and fade out at the edges (--------.:xxXXXXXXxx:.-------). If i move left or right, this led bar should follow.

After a certain time of inactivity, the bar should fade out.
This mode could also be used in a hallway or stairs.

umar14, 2019-07-25:
Hello everyone,

First of all thank you for this wonderful library. I have found it very useful. 

I have tried integrating [McLighting](https://github.com/toblum/McLighting) which uses [WS2812fx](https://github.com/kitesurfer1404/WS2812FX) to provide 50+ effects. If anyone is interested, you can clone and test my fork [here](https://github.com/umar14/Sonoff-Tasmota). Change `my_led_config.h` file accordingly. To control the effects using MQTT, send payload to topic `cmnd/sonoff`. To see results, subscribe to `stat/sonoff/LED`. Checkout [this](https://github.com/toblum/McLighting/wiki/WebSocket-API) for how to send payload. I don't have a sonoff so tested this with nodemcu board and seems to work. Don't know if it messes up other sonoff code. Hope this helps someone!

Regards,

thopd88, 2019-07-31:
Thanks @umar14 for the code, it works! Now I have beautiful led strip with lots of effects controlled by my mighty sonoff. Still need to improve the code thought, for example the led status does not save, and the default led color always turned on after power loss..

umar14, 2019-07-31:
@thopd88 Glad to see it works for you. You need SPIFFS to be able to store led config on reboot. Sonoff does not support SPIFFS. I tested it with nodemcu board (4M flash with SPIFFS) and it worked. But if you really need to save changes, you can compile with the default mode set to OFF so it boots up blank and then publish payload to sonoff with a retain flag. That way whatever mode or speed you publish, it will be republished if the sonoff reboots. I think.

ascillato2, 2019-10-28:
Changing the label **Enhancement** to **Feature Request** due to nobody is working on this feature at this moment. 

Besides, there are already complete mqtt solutions for led effects at: 

https://github.com/kitesurfer1404/WS2812FX
https://github.com/Aircoookie/WLED
https://github.com/Snipercaine/WLED-HomeAssistant

If someone is interested on working on this, please leave a comment. Otherwise the Stale Bot will close it. 

Thanks everyone for sharing theirs ideas.

mczeus, 2019-10-28:
I'm really interested in that. Unfortunately, I can not contribute much to the progress, but I would provide for testing

cruisinger, 2019-10-29:
Same for me. My programming skills are very limited, but I like blinking things and can only contribute some effects from my own project.

meingraham, 2019-10-29:
@bagobones PR would be welcome

ascillato2, 2019-11-01:
Closing this issue as nobody is working actively on this.

If someone wants to make a PR about this, please do it, and if need help on that, please just ask here or in the Tasmota Support Chat.

Adding the label **Requested Feature (Hold Over)** for future reference.

Thanks everyone for sharing their ideas. Very appreciated.

closed 2019-11-01 by ascillato2

a-roz, 2020-04-17:
https://github.com/a-roz/Tasmota/tree/ws2812fx

ozett, 2020-04-18:
wonderfull!

@a-roz 
maybe  the [readme.md](https://github.com/a-roz/Tasmota/tree/ws2812fx#commands) should clarify that the tasmota-fork shemes 13-68 
[image]

are the WS2812FX-[effects ](https://github.com/kitesurfer1404/WS2812FX#effects ) from 0 to 55 (=56 total) from the WS2812FX lib.
[image]

[image]

i wanted to check which effects are in the fork an i found no direct descrition on the fork´s readme.md, so i digged into this. maybe only a link to the names in the ws2812fx-effectslist makes it easier to lookup?

a-roz, 2020-04-18:
Mine 13..68 is the same of original library 0..55 with offset of 13 Tasmota embedded. If you need detaul about each of them, better you have a look inside library. "Xname" command result also has a fierld , contains effect name

FearNaBoinne, 2020-11-04:
Was this ever merged in to the main tree? (As the documentation at https://tasmota.github.io/docs/Commands/#scheme doesn't mention the extra schemes)

ascillato, 2020-11-04:
> Was this ever merged in to the main tree?

No, Sorry. No one had provided any PR for that.

FearNaBoinne, 2020-11-05:
So many good things get dropped by the wayside because people don't do PRs... If only I had time... But then again, if I did, I'd also have had time to work on the ADS1115 additional feature I want to implement...  😝

s-hadinger, 2020-11-05:
Find a motivated developer with spare time :)

FearNaBoinne, 2020-11-05:
Is that an offer?

Otherwise it's two sets of two words that typically do not combine:
"Motivated developer"
"Spare time"

But the intent is appreciated!  😁 

If nobody else does it, I hope to eventually be able to do it, but until then I will have to keep my fingers crossed someone will do it before I finally have some time!

KDMcMullan, 2020-12-02:
That's not mentioned in the previous issue. My supply is delivering 5.00V according to my multimeter. The supply is rated at 3A. The cable between my PSU and the LEDs is the manufacturer's cable and is only 20cm long.
I'm struggling for an explanation as to why I can flawlessly set any colour I like, and fade works perfectly when manually selected from the web interface: the only place fade doesn't work is from the schemes.