arendst/Tasmota #3482: sonoff pow r2

matthewjporter, 2018-08-14:
Hi, I load the firmware onto a sonoff pow r2 and can connect fine when in default mobde of sonoff basic module

I cannoect vial serial interface and setup the SSID and Password, connects fine to the WiFi and can manage on wifi or via TasmoAdmin in HomeAssistant

If I change the module to sonoff pow it still works as per normal and should

If i change to sonoff pow r2, it reboots as normal and can still do config via webpage but the serial connect ion is scambled unless I change the speed to 4800

I can no longeradd or manage via TasmoAdmin in HA

If i change back to a sonoff pow via the webinterface or serial and I still get garbage and cannot control in tasmoadmin

I can reset the config via the web, but then have to redo the config.

At the moment it is configured as a pow not a pow r2 and is working

Regards
Matthew

matthewjporter, 2018-08-14:
Defo a POW R2 V1.0, well, according to the label on it :)

I have flased it a few times now
[image]

arendst, 2018-08-14:
The R2 is using the serial interface as 4800 bps. Do not use the serial interface on a pow as it is used by the energy monitoring chip.

Frogmore42, 2018-08-14:
The POW R2 and POW use completely different measuring components.  The R2 uses a chip that has a serial interface.  This means you can NOT use the serial interface for anything else.  If you have a serial adapter connected, it will compete with energy chip and neither will be happy about it.

You said you had it configured as a POW and it was working.  They both use the same GPIO for the relay and red led, but the energy chips are different.  If you are getting accurate energy readings from a POW R2 that is configured as a POW, I have no explanation as to how that is possible.

Your picture certainly seems to confirm that you have an R2, so I would configure the WiFi from serial, with the module set as a basic and then remove the serial adapter and configure it from the web interface as a R2.

matthewjporter, 2018-08-14:
Sorry, should have said that I have no power connected yet, just the serial for the flash and adding to WiFi then the web ui is and linking to TasmoAdmin in Hassio is working as basic or pow

I will hook up to power directly now and advise what the results are when changing to pow r2

The bit i really want working is the linking to TasmoAdmin

matthewjporter, 2018-08-14:
Just thought I would also mention I think tasmota is the best firmware I have found for integrating my Sonoff devices with Homeassistant. I have 15 sonoff th, 5 ifan, 1 POW and now setting up 5 POW R2's and only growing larger as I will soon be migrating my Wemo's D1 Mini/Pro projects over - Keep up the good work

Ok back to my issue, So i have banged up a quick test rig with my soldering iron on a plug socket with the POW R2 as the middle man
[image]

The POW R2 is showing my soldering iron as drawing a steady 25w
[image]

This is all well and good as expected.

Ok, I understand the serial will not work anymore so first issue resolved, well it does but at 4800 :) and yes I know not to connect serial and AC at the same time

Here is the problem I face, I can get to the POW R2 UI over my WiFi as you can see from above and is displaying voltage, current and power etc as expected and can mage that way

The problem I get is that as soon as I set this to a POW R2 from a sonoff basic I can no longer manage in TasmoAdmin in my Homeassistant setup and get the error below, however Tasmoadmin does show th power stats and switch status as per second shot below
[image]

[image]

Note the unit called sonoffpow-01 is a normal POW and I can manage it fine, The one with the issue is labeled as sonoffpow-02 and I just cannot access the config/edit screens on TasmoAdmin

I am not sure if this maybe an issue with TasmoAdmin not being able to access the POW R2 correctly will be posting on that page and see if I get a response

matthewjporter, 2018-08-14:
Update: Seems that this is a timeout issue as I can very randomly and very far between actually get to the config page

I have tried the unit on different IP's but am sure I have no conflicts as it is a DHCP reserved address

All my other units are working fine

Frogmore42, 2018-08-14:
You did not include the Status 0 information, which would indicate which core you are using.  Core 2.3.0 is the current recommended one.  2.4.2 is the newest, but does not have a lot of people using it yet.  2.4.0 is bad .  2.4.1 doesn't work for a lot of people.

matthewjporter, 2018-08-14:
Hi,

here is the current status, the Core is listed as 2.3.0

19:39:51 MQT: stat/sonoffpow-02/STATUS = {"Status":{"Module":43,"FriendlyName":["sonoffpow-02"],"Topic":"sonoffpow-02","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
19:39:51 MQT: stat/sonoffpow-02/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:54:03","StartupUTC":"2018-08-14T17:45:48","Sleep":0,"BootCount":22,"SaveCount":42,"SaveAddress":"FA000"}}
19:39:51 MQT: stat/sonoffpow-02/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-31T17:59:09","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:39:51 MQT: stat/sonoffpow-02/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxxxxxxxxx",""],"TelePeriod":300,"SetOption":["00008029","55818000"]}}
19:39:51 MQT: stat/sonoffpow-02/STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","03B6179E","00000000"]}}
19:39:51 MQT: stat/sonoffpow-02/STATUS5 = {"StatusNET":{"Hostname":"sonoffpow-02","IPAddress":"xxxxxxxxxxxxx","Gateway":"xxxxxxxxxxxxx","Subnetmask":"xxxxxxxxxxxxx","DNSServer":"xxxxxxxxxxxx","Mac":"xxxxxxxxxxxxxxx","Webserver":2,"WifiConfig":3}}
19:39:51 MQT: stat/sonoffpow-02/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxxxxxxxxxx","MqttPort":1883,"MqttClientMask":"sonoffpow-02","MqttClient":"sonoffpow-02","MqttUser":"xxxxxxxxxxxxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:39:51 MQT: stat/sonoffpow-02/STATUS7 = {"StatusTIM":{"UTC":"Tue Aug 14 18:39:51 2018","Local":"Tue Aug 14 19:39:51 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:41","Sunset":"20:08"}}
19:39:51 MQT: stat/sonoffpow-02/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:39:51 MQT: stat/sonoffpow-02/STATUS10 = {"StatusSNS":{"Time":"2018-08-14T19:39:51","ENERGY":{"Total":0.031,"Yesterday":0.000,"Today":0.031,"Power":0,"Factor":0.00,"Voltage":222,"Current":0.000}}}
19:39:51 MQT: stat/sonoffpow-02/STATUS11 = {"StatusSTS":{"Time":"2018-08-14T19:39:51","Uptime":"0T00:54:03","Vcc":3.173,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxxxxxxxxx","RSSI":44,"APMac":"xxxxxxxxxxxxxx"}}}

Frogmore42, 2018-08-15:
Well, that looks okay.  Can you access the web console from a browser?

I haven't seen not heard of people having issues with the web console locally.

Do you have Hue emulation turned on.  I have heard that can cause issues on some networks.

ascillato2, 2018-08-21:
Hi,

Any updates on this?

matthewjporter, 2018-08-21:
Sorry, I have been away for a few days

I have erased and reflased a few times and was repeatable each time and had basically decided as long as it was sending the data via mqtt and I could get direct onto the WebGUI I would for go the TasmoAdmin configuration portion

And since you reminded me to check, it now seems to be working which I can only assume is after the update that was released to TasmoAdmin a few days back (https://github.com/hassio-addons/addon-tasmoadmin/releases)

The only thing I can guess is the updates in v0.5.0 resolved the issue

Thanks Guys for all the help

closed 2018-08-21 by matthewjporter