arendst/Tasmota #3484: Better explain what is Sonoff Pow Factor

afaucogney, 2018-08-14:
Hello,

 It Could be classified as documentation bug, as I do not understand what is the Pow Factor:
- doc says :  The ratio of the real power flowing to the load to  the apparent power in the circuit.

Could someone give me more info and a real exemple.

Thanks

arendst, 2018-08-14:
Search using google.

Jason2866, 2018-08-14:
https://en.wikipedia.org/wiki/Power_factor

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2

afaucogney, 2018-08-28:
@arendst @Jason2866 thanks for answering. Now I get it ! I apologize for having been seen as stupid, but..., it was not obvious that we talk about "Power Factor", the doc write just "Factor", as "Current", "Energy" ..., so the Power could be understood as the "Power" of the "Sonoff Power".
Additionally, first : I m not a native English speaker, and second : in France (I have some level in electrical science), we talk more about "Cos Phi" than "Power Factor".