arendst/Tasmota #3485: Thermostat Example Rule

pcpaulo, 2018-08-14:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

pcpaulo, 2018-08-14:
I am using  the Thermostat Example and i am getting MQTT answers every second. There is a way to inhibit the MQTT message for rules?

Jason2866, 2018-08-14:
Example is in wiki under rules

Jason2866, 2018-08-14:
Look here
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#4-usage-of-one-shot-once

pcpaulo, 2018-08-14:
this example explain how to publish and my objective is not received the publish MQTT command

Jason2866, 2018-08-14:
With one-shot you get only ONE mqtt message send from Tasmota device.
I understand you that you dont want many messages send from the thermostat.

pcpaulo, 2018-08-14:
rule 4 turn off one-shot rule ; how to turn on ?

Jason2866, 2018-08-14:
See wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management

ascillato2, 2018-08-18:
Hi,

Closing this issue as your questions has been addressed. Ask to reopen this issue if you need it. Thanks

closed 2018-08-18 by ascillato2