arendst/Tasmota #3487: TM1638, all buttons same topic

wunderbaum, 2018-08-14:
I installed the newest version to try out the tm1638 function. Unfortunately, I always get the same topic on console and mqtt whatever button I press.
I can set different switchmodes to each button, but the topic is always stat/sonoff/POWER
How can I change this?

**Describe the bug**

_A clear and concise description of what the bug is._

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X] Searched the problem in issues and in the wiki
- [X ] Hardware used : Wemos D1 
- [ X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
20:26:14 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-08-14:
See #3214

wunderbaum, 2018-08-15:
Hi,
what I tried on the console to give each of the 8 buttons of the tm1638 a different topic. What I did:
```
switchtopic1 2 bla
switchtopic2 2 blubb
switchtopic3 2 lala
```
Console responses:
```
18:42:18 MQT: stat/tm1638/RESULT = {"SwitchTopic":"2_bla"}
18:43:31 MQT: stat/tm1638/RESULT = {"SwitchTopic":"2_blubb"}
18:44:11 MQT: stat/tm1638/RESULT = {"SwitchTopic":"2_lala"}
```
and the behaviours of the buttons
```
switchmode1 3
switchmode2 1
switchmode3 4
...
```

So, I expected to get now different topics published. But its always the last given topic for every button.
`18:45:06 MQT: cmnd/2_lala/POWER = TOGGLE`
I think I am doing something completely wrong, what do I need to do to see WHICH button on the tm1638 is pressed?

Jason2866, 2018-08-15:
Take a look:
`18:51:08 MQT: cmnd/sonoff-1A2AFC/POWER1 = ON
18:51:08 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER":"ON"}
18:51:08 MQT: stat/sonoff-1A2AFC/POWER = ON
18:51:10 MQT: cmnd/sonoff-1A2AFC/POWER1 = OFF
18:51:10 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER":"OFF"}
18:51:10 MQT: stat/sonoff-1A2AFC/POWER = OFF
18:51:13 MQT: cmnd/sonoff-1A2AFC/POWER2 = ON
18:51:13 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:51:13 MQT: cmnd/sonoff-1A2AFC/POWER2 = OFF
18:51:13 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:51:26 CMD: switchtopic1
18:51:26 MQT: stat/sonoff-1A2AFC/RESULT = {"SwitchTopic":"sonoff-1A2AFC"}
18:51:37 CMD: switchtopic2
18:51:37 MQT: stat/sonoff-1A2AFC/RESULT = {"SwitchTopic":"sonoff-1A2AFC"}
18:51:44 CMD: switchmode1
18:51:44 MQT: stat/sonoff-1A2AFC/RESULT = {"SwitchMode1":4}
18:51:50 CMD: switchmode2
18:51:50 MQT: stat/sonoff-1A2AFC/RESULT = {"SwitchMode2":2}
18:52:04 CMD: switchmode3
18:52:04 MQT: stat/sonoff-1A2AFC/RESULT = {"SwitchMode3":2}
18:52:11 MQT: tele/sonoff-1A2AFC/STATE = {"Time":"2018-08-15T18:52:11","Uptime":"0T00:46:02","Vcc":3.040,"POWER":"OFF","Dimmer":12,"Color":"1E1E1E","HSBColor":"0,0,12","Channel":[12,12,12],"Scheme":11,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":70,"APMac":"00:A0:57:2A:BD:19"}}
18:52:11 MQT: tele/sonoff-1A2AFC/SENSOR = {"Time":"2018-08-15T18:52:11","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","Switch5":"OFF","Switch6":"OFF","Switch7":"OFF","Switch8":"OFF"}`

If i press Button1 (switchmode1 4 = toggle) the connected WS2812 can be switched On/Off

Jason2866, 2018-08-15:
18:54:54 CMD: switchtopic
18:54:54 MQT: stat/sonoff-1A2AFC/RESULT = {"SwitchTopic":"sonoff-1A2AFC"}
18:55:46 MQT: cmnd/sonoff-1A2AFC/POWER1 = ON
18:55:46 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER":"ON"}
18:55:46 MQT: stat/sonoff-1A2AFC/POWER = ON
18:55:47 MQT: cmnd/sonoff-1A2AFC/POWER1 = OFF
18:55:47 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER":"OFF"}
18:55:47 MQT: stat/sonoff-1A2AFC/POWER = OFF
18:55:49 MQT: cmnd/sonoff-1A2AFC/POWER2 = ON
18:55:49 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:49 MQT: cmnd/sonoff-1A2AFC/POWER2 = OFF
18:55:49 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:51 MQT: cmnd/sonoff-1A2AFC/POWER2 = ON
18:55:51 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:52 MQT: cmnd/sonoff-1A2AFC/POWER2 = OFF
18:55:52 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:53 MQT: cmnd/sonoff-1A2AFC/POWER3 = ON
18:55:53 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:53 MQT: cmnd/sonoff-1A2AFC/POWER3 = OFF
18:55:53 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:54 MQT: cmnd/sonoff-1A2AFC/POWER4 = ON
18:55:54 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:54 MQT: cmnd/sonoff-1A2AFC/POWER4 = OFF
18:55:54 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:54 MQT: cmnd/sonoff-1A2AFC/POWER5 = ON
18:55:54 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:55 MQT: cmnd/sonoff-1A2AFC/POWER5 = OFF
18:55:55 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:55 MQT: cmnd/sonoff-1A2AFC/POWER6 = ON
18:55:55 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:55 MQT: cmnd/sonoff-1A2AFC/POWER6 = OFF
18:55:55 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:55 MQT: cmnd/sonoff-1A2AFC/POWER7 = ON
18:55:55 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:56 MQT: cmnd/sonoff-1A2AFC/POWER7 = OFF
18:55:56 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:56 MQT: cmnd/sonoff-1A2AFC/POWER8 = ON
18:55:56 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:56 MQT: cmnd/sonoff-1A2AFC/POWER8 = OFF
18:55:56 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:02 MQT: cmnd/sonoff-1A2AFC/POWER1 = ON
18:56:02 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER":"ON"}
18:56:02 MQT: stat/sonoff-1A2AFC/POWER = ON
18:56:03 MQT: cmnd/sonoff-1A2AFC/POWER1 = OFF
18:56:03 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER":"OFF"}
18:56:03 MQT: stat/sonoff-1A2AFC/POWER = OFF
18:56:04 MQT: cmnd/sonoff-1A2AFC/POWER2 = ON
18:56:04 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:04 MQT: cmnd/sonoff-1A2AFC/POWER2 = OFF
18:56:04 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:05 MQT: cmnd/sonoff-1A2AFC/POWER2 = ON
18:56:05 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:06 MQT: cmnd/sonoff-1A2AFC/POWER2 = OFF
18:56:06 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:07 MQT: cmnd/sonoff-1A2AFC/POWER3 = ON
18:56:07 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:08 MQT: cmnd/sonoff-1A2AFC/POWER3 = OFF
18:56:08 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:10 MQT: cmnd/sonoff-1A2AFC/POWER3 = ON
18:56:10 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:13 MQT: cmnd/sonoff-1A2AFC/POWER3 = OFF
18:56:13 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:14 MQT: cmnd/sonoff-1A2AFC/POWER4 = ON
18:56:14 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:16 MQT: cmnd/sonoff-1A2AFC/POWER4 = OFF
18:56:16 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:17 MQT: cmnd/sonoff-1A2AFC/POWER5 = ON
18:56:17 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:18 MQT: cmnd/sonoff-1A2AFC/POWER5 = OFF
18:56:18 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:19 MQT: cmnd/sonoff-1A2AFC/POWER6 = ON
18:56:19 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:20 MQT: cmnd/sonoff-1A2AFC/POWER6 = OFF
18:56:20 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:20 MQT: cmnd/sonoff-1A2AFC/POWER7 = ON
18:56:20 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:22 MQT: cmnd/sonoff-1A2AFC/POWER7 = OFF
18:56:22 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:22 MQT: cmnd/sonoff-1A2AFC/POWER8 = ON
18:56:22 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:56:23 MQT: cmnd/sonoff-1A2AFC/POWER8 = OFF
18:56:23 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}

Jason2866, 2018-08-15:
You get via mqtt  cmnd/xxxx/PowerX = ON / OFF
If a switchable device is connected (in my example an WS2812) you get a Result message.
with Option26 1  you get for first device this message:
19:01:08 CMD: setoption26 1
19:01:08 MQT: stat/sonoff-1A2AFC/RESULT = {"SetOption26":"ON"}
19:01:11 MQT: cmnd/sonoff-1A2AFC/POWER1 = ON
19:01:11 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER1":"ON"}
19:01:11 MQT: stat/sonoff-1A2AFC/POWER1 = ON
19:01:15 MQT: cmnd/sonoff-1A2AFC/POWER1 = OFF
19:01:15 MQT: stat/sonoff-1A2AFC/RESULT = {"POWER1":"OFF"}
19:01:15 MQT: stat/sonoff-1A2AFC/POWER1 = OFF

SetOption26 0 / off -> (default) Keep using POWER without postfix for single power devices
SetOption26 1 / on -> Add postfix to all POWER messages

wunderbaum, 2018-08-16:
Sounds more complicated than expected. Puuh :-)
What I like to have is different MQTT-publishes per button on the TM1638.

emontnemery, 2018-08-23:
@wunderbaum The difference between "switch" and "button" in Tasmota is that:
- A "Switch" is an electrical device that can be set to ON or OFF, and remains there (like a mechanical light switch, for example).
- A button is like a push button, you press it, and when you release it springs back. For example a touch button on the Sonoff T1.

You do get different MQTT-published per button, for example like this:
23:07:29 CMD: buttontopic
23:07:29 MQT: tasmota/sonoff_A12D7C/stat/RESULT = {"ButtonTopic":"sonoff_BTN_%06X"}

Press button 1 will publish:
tasmota/sonoff_BTN_A12D7C/cmnd/POWER1 = TOGGLE
Press button 2 will publishes:
tasmota/sonoff_BTN_A12D7C/cmnd/POWER2 = TOGGLE

ascillato2, 2018-08-24:
@emontnemery

> Regarding "button" vs "switch", is it described anywhere on the wiki?

No.

Please, can you add a note in the wiki at that page? Thanks.

wunderbaum, 2018-08-25:
Unfortunately with the tm1638 the POWER1 , POWER2,... is missing in my setup.

ascillato, 2018-08-29:
Hi,

Have you managed to solve your issue?

wunderbaum, 2018-08-29:
Unfortunately, not. I will make a complete new setup these days to check what I am doing wrong.

wunderbaum, 2018-08-29:
Hello,
I installed a complete new wemos and connected the tm1638.
Whatever button I press on the module, I always get only:
17:42:55 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
17:42:55 MQT: stat/sonoff/POWER = OFF``

ascillato, 2018-08-29:
Hi,

Have you set more than 1 relay?

wunderbaum, 2018-08-30:
Good morning!
No, I have not set any relay because the board I am using has no relay. Its this [model](https://www.aliexpress.com/item/TM1638-Module-Key-Display-New-8-Bit-Digital-LED-Tube-8-Bit/32883048249.html?spm=2114.search0104.3.2.1ec57f15S5Exfi&ws_ab_test=searchweb0_0,searchweb201602_5_10320_10065_10068_10843_10547_10059_10548_10696_100031_10319_10084_10083_10103_10618_10304_10307_10820_10821_10302,searchweb201603_16,ppcSwitch_5&algo_expid=009bc2ea-a02a-489a-b6f0-3ef39f08f5ed-0&algo_pvid=009bc2ea-a02a-489a-b6f0-3ef39f08f5ed&transAbTest=ae803_1&priceBeautifyAB=0):

Jason2866, 2018-08-30:
Good morning @wunderbaum 

Try following set a virtual relais to a free not used GPIO. This means nothing real connected.
SetOption26 1  // on -> Add postfix to all POWER messages
What do you have set for Switchmode1 Switchmode2 ... to Switchmode8 ?

On switches you have nothing connected to switch you will get:

18:55:49 MQT: cmnd/sonoff-1A2AFC/POWER2 = ON
18:55:49 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:49 MQT: cmnd/sonoff-1A2AFC/POWER2 = OFF
18:55:49 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:51 MQT: cmnd/sonoff-1A2AFC/POWER2 = ON
18:55:51 MQT: stat/sonoff-1A2AFC/RESULT = {"Command":"Unknown"}
18:55:52 MQT: cmnd/sonoff-1A2AFC/POWER2 = OFF

wunderbaum, 2018-08-30:
Hello!
Now I defined 8 relay to unused gpio. Now I get the POWER1..POWER8 messages! Halleluja!
But this is really a waste of GPIOs.

Jason2866, 2018-08-30:
It should work if one relais is defined. So it does for me.

wunderbaum, 2018-08-30:
Does not work here. Hmpf. Then I only get POWER1m whatever button I press. I tried both, relay1 and relay1i. 

switchmode1 is 2. Setoption26 is 1.

ascillato2, 2018-09-03:
Hi, Have you tried rules as the examples in the wiki?

wunderbaum, 2018-09-03:
Hello!
Thanks for your help, you are really patient with me! So far I haven't tried rules.  I will look at it.
I am now in small progress for an own program as I only need to show something on the display and get the state of the buttons. Good brain training :-)

ascillato2, 2018-09-04:
Ok,

As you change Tasmota to your own program, this issue is no longer necessary. Closing it.

If you need further help, please, ask to reopen this issue. Thanks.

closed 2018-09-04 by ascillato2