arendst/Tasmota #3488: Sonoff SC and BME280

ilvenos, 2018-08-14:
Hi,
Is it possible to connect BME280 to Sonoff SC instead of TH11?
I see only one free GPIO when I chose Sonoff SC.

Frogmore42, 2018-08-15:
You would have to find the other GPIOs and modify the template.  Unless you are really good with a soldering iron, it would be easier to get another device.

Frogmore42, 2018-08-15:
The TH11 is connected to the other MCU, so you can't replace it.

ilvenos, 2018-08-15:
@mike2nl  could you show me Your modules configuration in tasmota?
When I chosen Sonoff SC I have only one GPIO (GPIO2).
if I understand you correctly, you didnt chose Sonoff SC from tasmota settings?

mike2nl, 2018-08-17:
@ilvenos 
no issue but you have to wait that i'm at home. Round the afternoon i'm
back and i will made some screenshots....

ascillato, 2018-08-23:
@ilvenos

Hi,

Have you managed to solve your issue?

ascillato2, 2018-08-31:
@mike2nl help please

ascillato2, 2018-09-03:
Hi,

@ilvenos

Please, try to contact mike in discord chat.

As this issue is on HOLD since 17 ago, I'm closing it temporarily until you have time to continue with it. Please, as soon as you have more information/questions, ask to reopen this issue and we gladly continue with it. Thanks.

closed 2018-09-03 by ascillato2