arendst/Tasmota #3490: Sonoff S20 flashing upload is canceled

Smarti15, 2018-08-15:
Hi I'm trying to flash a Sonoff S20_EU_V1.3. Unfortunately, I always get the following message when uploading:

Configuring upload protocol ...
Looking for upload port ...
Use manually specified: COM5
Uploading .pioenvs \ sonoff-DE \ firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed

error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 20.88 seconds
Environment sonoff-DE [ERROR]
 [ERROR] Took 20.88 seconds

Do you have any idea what that could be?

curzon01, 2018-08-15:
pls read [Wiki FAQ - Flashing Issues](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#flashing-issues)

curzon01, 2018-08-15:
If you are not sure, try it
- reverse TX/RX lines, this will not damaged anything
- try holding button during the whole flash procedure incl. boot of device, this will not damaged anything

Smarti15, 2018-08-15:
I found the mistake. The cables that came with the CP2102 were all broken. I replaced it and immediately everything worked out. :)

curzon01, 2018-08-15:
**(Please, remember to close the issue when the problem has been addressed)**

closed 2018-08-18 by ascillato2