arendst/Tasmota #3493: Need a command to turn on/off any specific user IO

qingz2004, 2018-08-15:
**Have you look for this feature in other issues and in the wiki?**
Yes.
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
No.
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
A simple command to set a specific user IO to 0 or 1.
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
Need to toggle some user IO when relay is controlled by the serial port through a MCU.
**Additional context**
_Add any other context or screenshots about the feature request here._
I can not set a user IO to control relay when there is another relay is controlled by the serial port.
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-08-15:
Define the Gpio as relay. You can control it easily this way.
I dont follow this:
I can not set a user IO to control relay when there is another relay is controlled by the serial port.
What brings you to this conclusion?

qingz2004, 2018-08-15:
@Jason2866 Like Sonoff Dual, the onboard relays are controlled by a micro controller through the serial port. You can not toggle GPIO 4 even you set it as a third relay.

Jason2866, 2018-08-15:
This additional MCU is NOT under control from Tasmota. It has stock firmware inside.
The functionality of the reactions of the MCU was done via reverse engeneering.
There will no more options for this part of the device... or you can do it ;-)
If you want to control Gpios the have to be from the ESP where Tasmota is running.
You can connect here extension devices.

qingz2004, 2018-08-15:
You can try to set a device as Sonoff Dual, and configure GPIO4 as a relay, after reboot, there are still only two relays to toggle from GUI. And if you try to send Power command from console, the command "Power3 on" is "Unknown".

Jason2866, 2018-08-15:
Have to try this setup. In general Gpio4 should work for a 3rd relais!
If not this is a issue.

qingz2004, 2018-08-15:
Sonoff Dual is just an example. With any module which uses serial port to control relay, you can not toggle the user GPIOs directly from Tasmota.

wongnam, 2018-08-16:
Buy Sonoff Dual R2
Work perfect with Tasmota

qingz2004, 2018-08-24:
Any progress?

ascillato, 2018-08-27:
Hi,

Issue confirmed. Feature not supported.

When choosing a device that uses the serial port to control relays, more relays can not be added.

qingz2004, 2018-09-20:
Can't this be fixed or added?
All what I need is to toggle a specific IO with a simple web console or MQTT command when the serial port is used in this device.

Jason2866, 2018-09-20:
@qingz2004 
At https://discord.gg/Ks2Kzd4 you may find help for your answer

ascillato, 2018-09-20:
Hi,

Up to now, this feature is not supported by Tasmota _(If serial port is used to control relays, more relays can not be added.)_

This open issue has the enhancement label to be taken into account. With this label the Bot will not close it. Don't worry.

I don't know when, nor who, but may be someone will take this enhancement and start working on it.

KPK75, 2018-09-21:
Use generic module, define Relays ( some of them fake ) and use rules! You can have direct control via GPIO and indirect via MCU using rules and SerialSend comand ( if you know protocol ). Search for LCTech relay as example.

qingz2004, 2018-09-21:
@KPK75 thanks! I will try.

andrethomas, 2018-10-02:
One option could be to add an MCP23008 or MCP23017 if the I2C pins are accessible - Then you can have additional outputs via I2C... but it is not controlled with POWER command's though

qingz2004, 2018-10-02:
@andrethomas Thanks! I know this work around, but it's not my option here since there is no space in the product. And it not worth to add a module just for one GPIO while I have a lot of spare IO from the ESP8266 itself. The problem is we should not have this limitation. Unfortunately I'm not good in software programming.

andrethomas, 2018-10-02:
The only alternative is to figure out the protocol used by the serial interface and then configure as a generic device and use sendserial commands to control the relay controller chip.

qingz2004, 2018-10-02:
Yes, that should work, but I lost the GUI from its web interface.

andrethomas, 2018-10-02:
Well, yes you will loose the power1 and power2 from the web ui in its original form, but you can just create relays on unused pins and then use rules to use them to as triggers to send serial data.

qingz2004, 2018-10-02:
I also need to read the serial port to get the status.

andrethomas, 2018-10-02:
If you maintain the status in a VAR then you don't need to read it I guess

qingz2004, 2018-10-02:
No, the status could change by external buttons.

andrethomas, 2018-10-02:
No solution for this one then, I guess.

ascillato2, 2018-10-02:
I think there is not a workaround for this. I think we need to fix this in the code.

andrethomas, 2018-10-02:
@ascillato Yeah, agree - guess we'll have to dig in the code and add some command for using unconfigured GPIO's - I know there is a GPIO command but it currently only gives control to ones defined in the template, both of which in this case seems to cause a reset loop.

ascillato2, 2018-11-16:
This requested feature is not being addressed by anyone. Added the label hold over for future reference. Thanks everyone for sharing theirs ideas.

closed 2018-11-16 by ascillato2