arendst/Tasmota #3494: IRSEND HEX not being recognized by TV

hubertbanas, 2018-08-15:
Samsung Volume Down

```
23:36:13 MQT: tele/wemos1/RESULT = {"IrReceived":{"Protocol":"SAMSUNG","Bits":32,"Data":"E0E0D02F"}}
```

Nothing happens when sending HEX as follow:

```
21:27:27 CMD: irsend {"Protocol":"SAMSUNG","Bits":32,"Data":"E0E0D02F"}
21:27:27 RSL: Received Topic /irsend, Data Size 50, Data {"Protocol":"SAMSUNG","Bits":32,"Data":"E0E0D02F"}
21:27:27 RSL: Group 0, Index 1, Command IRSEND, Data {"Protocol":"SAMSUNG","Bits":32,"Data":"E0E0D02F"}
21:27:27 IRS: protocol_text SAMSUNG, protocol SAMSUNG, bits 32, data 0 (0x0), protocol_code 7
21:27:28 MQT: stat/wemos1/RESULT = {"IRSend":"Done"}
21:27:28 IRR: RawLen 68, Bits 34, Value 3FFF60FC, Decode -675632700
```

Volume goes DOWN when sending DEC as follow:

```
21:28:11 CMD: irsend {"Protocol":"SAMSUNG","Bits":32,"Data":"3772829743"}
21:28:11 RSL: Received Topic /irsend, Data Size 52, Data {"Protocol":"SAMSUNG","Bits":32,"Data":"3772829743"}
21:28:11 RSL: Group 0, Index 1, Command IRSEND, Data {"Protocol":"SAMSUNG","Bits":32,"Data":"3772829743"}
21:28:11 IRS: protocol_text SAMSUNG, protocol SAMSUNG, bits 32, data 3772829743 (0xE0E0D02F), protocol_code 7
21:28:11 MQT: stat/wemos1/RESULT = {"IRSend":"Done"}
21:28:11 IRR: RawLen 68, Bits 32, Value 00000001, Decode -522137553
```

Same behavior with Volume Up, Power Toggle, Mute Toggle, etc

ascillato2, 2018-08-25:
Hi,

From the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#irremote), if using hex: 

> "Data": 0x1..0xFFFFFFFF} | data is the data frame as 32 bit hexadecimal

You need to use **_0x_** and also use `SetOption29`

Please, try that and if it works close this issue. Thanks.

hubertbanas, 2018-08-26:
Thank you for pointing out the wiki. Indeed adding **0x** did the trick.

```
09:48:24 CMD: irsend {"Protocol":"SAMSUNG","Bits":32,"Data":"0xE0E0D02F"}
09:48:24 RSL: Received Topic /irsend, Data Size 52, Data {"Protocol":"SAMSUNG","Bits":32,"Data":"0xE0E0D02F"}
09:48:24 RSL: Group 0, Index 1, Command IRSEND, Data {"Protocol":"SAMSUNG","Bits":32,"Data":"0xE0E0D02F"}
09:48:24 IRS: protocol_text SAMSUNG, protocol SAMSUNG, bits 32, data 3772829743 (0xE0E0D02F), protocol_code 7
09:48:24 MQT: stat/wemos1/RESULT = {"IRSend":"Done"}
09:48:24 IRR: RawLen 68, Bits 32, Value 00000001, Decode -522137553
```

closed 2018-08-26 by hubertbanas