arendst/Tasmota #3496: 10% of the OTA firmware uploads will fail, unless you reboot sonoff after OTA upgrade

lobocobra, 2018-08-15:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : SONOFF BASIC, CH4, CH4-pro, POW
- [ ] Provide the output of command ``status 0`` : 
```
stat/sonoff/STATUS = {"Status":{"Module":7,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
```

I am currently extending the basic firmware with some additional functions in order to operate shutters, oil heatings and water dispenser in the garden. Doing that I had to test the firmware several 100 times and 10% of the uploads failed, ending with an serial upload.
=> Did anyone else make a similar observation? 

When I do a serial upload or I reboot the SONOFF before the next OTA upload, I have 100% success rate.
=> I would like to understand if it is worth the time if I start to look into the OTA section, what might cause this issue.

reloxx13, 2018-08-16:
this is not a status 0 response, pls post a complete status 0 response.

ascillato, 2018-08-26:
@lobocobra 

When you said reboot, are you meaning to reboot using the software (web button, console command, etc) or disconnecting power and connecting again?

lobocobra, 2018-08-26:
Yes I mean that I just have to do a software reboot. I do not need to do a hard - reset.
=> The work around works fine for me, so I only would invest time, if others see it as a problem too. 

But as there are not many doing 100 uploads a day, I guess  I gonna close the topic soon.

lobocobra, 2018-08-28:
As it seems this seems not to be a problem for a big part of the population. 
=> I will close the issue.

In case you have occasionally issues to make an OTA, you simply need to reboot by software the device after your OTA, in order to prevent, that you might have an issue on your next OTA.

closed 2018-08-28 by lobocobra