arendst/Tasmota #3499: Web Server Stops Responding Randomly - Sonoff Tasmota 5.14.0 by Theo Arends

lazers378, 2018-08-16:
**Describe the bug**
I am on version 5.14 and sometimes web server stops responding and sometimes it starts to server garbage HTML. The issue is totally random and I am not sure how to resolve. Can anyone help in this regard. I have SonOff Basic.
_A clear and concise description of what the bug is._

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : SonOff Basic with Tasmota 5.14.0
- [x] Provide the output of command ``status 0`` :
```
00:04:43 MQT: stat/PorchLight/STATUS = {"Status":{"Module":1,"FriendlyName":["Bed"],"Topic":"PorchLight","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:04:43 MQT: stat/PorchLight/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:04:26","StartupUTC":"","Sleep":0,"BootCount":1520,"SaveCount":2410,"SaveAddress":"FA000"}}
00:04:43 MQT: stat/PorchLight/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-28T18:28:32","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:04:43 MQT: stat/PorchLight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHome","SmartHome"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:04:43 MQT: stat/PorchLight/STATUS4 = {"StatusMEM":{"ProgramSize":548,"Free":452,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:04:43 MQT: stat/PorchLight/STATUS5 = {"StatusNET":{"Hostname":"Sonoff1","IPAddress":"192.168.100.102","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"EC:FA:BC:86:A1:35","Webserver":2,"WifiConfig":4}}
00:04:43 MQT: stat/PorchLight/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.111","MqttPort":1883,"MqttClientMask":"PorchLight","MqttClient":"PorchLight","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:04:43 MQT: stat/PorchLight/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:04:43 1970","Local":"Thu Jan 01 00:04:43 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":5,"Sunrise":"02:01","Sunset":"12:09"}}
00:04:43 MQT: stat/PorchLight/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:04:43"}}
00:04:43 MQT: stat/PorchLight/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:04:43","Uptime":"0T00:04:26","Vcc":3.370,"POWER":"ON","Wifi":{"AP":2,"SSId":"SmartHome","RSSI":62,"APMac":"68:89:C1:69:4E:14"}}}
00:04:47 MQT: tele/PorchLight/STATE = {"Time":"1970-01-01T00:04:47","Uptime":"0T00:04:30","Vcc":3.375,"POWER":"ON","Wifi":{"AP":2,"SSId":"SmartHome","RSSI":62,"APMac":"68:89:C1:69:4E:14"}}
```

**To Reproduce**
The issue is random. During the time the device is successfully responding in openhab and mqtt client and I am able to turn it on and off.

**Expected behavior**
It should display web server as device is connected with RSSI of over 60. Just now I kept on trying and it finally showed console after 15 retries.

lazers378, 2018-08-17:
I am still new to all this so please a bit more help. How do I use Core 2.3.0. What version do I flash to get it ?
Can it be done OTA also ?

Frogmore42, 2018-08-17:
If you are flashing a binary, you can use the latest one.  It is built with core 2.3.0.
Please read the wiki and the other issues related to core 2.3.0
https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade
Yes, you can do this OTA, but you will need to do the two step process.  Upload the minimal version and then the full version.

Since sometimes OTA does not work well, it is possible you will need to learn to use the serial flash method.

lazers378, 2018-08-17:
Thanks, one last thing. I think I figured 2.3.0. I was setting up Arduino on my new laptop and during the process I saw version 2.4.0 installed and now I installed 2.3.0 while setting up board manager.
I previously flashed using the USB method and will do the same again. Just have to unplug the sonoff from the light. I have the source and using that to flash. So I would just plug the sonoff using that cable and flash the compiled version again just like I did last time ?

lazers378, 2018-08-17:
Ok I reflashed it using core 2.3.0. Status is posted below, please check if it seems fine now. Meanwhile I am testing it
15:45:28 MQT: stat/sonoff1/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:45:29 MQT: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:09:27","StartupUTC":"2018-08-17T10:36:01","Sleep":0,"BootCount":4,"SaveCount":13,"SaveAddress":"F7000"}}
15:45:29 MQT: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-08-17T13:14:27","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:45:29 MQT: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartHome",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
15:45:29 MQT: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":408,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
15:45:29 MQT: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff2","IPAddress":"192.168.100.118","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"68:C6:3A:A3:5B:36","Webserver":2,"WifiConfig":4}}
15:45:29 MQT: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A35B36","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:45:29 MQT: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 17 10:45:29 2018","Local":"Fri Aug 17 15:45:29 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":5,"Sunrise":"09:45","Sunset":"00:02"}}
15:45:29 MQT: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2018-08-17T15:45:29"}}
15:45:29 MQT: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2018-08-17T15:45:29","Uptime":"0T00:09:28","Vcc":3.206,"POWER":"ON","Wifi":{"AP":1,"SSId":"SmartHome","RSSI":74,"APMac":"68:89:C1:69:4E:14"}}}

lazers378, 2018-08-17:
However, now my router is showing it as connected but the web server page still randomly shows sometimes and sometimes it says unable to find the page like sonoff is disconnected. However, on my router client list it is shown as online

lazers378, 2018-08-17:
Ok now I restarted the sonoff and it is connected. It seems to be working fine, timer worked correctly. No garbage served by web server but still randomly sometimes the web server does not respond (while being connected) and I have to keep refreshing and ultimately it serves the page.
My experience with Sonoff has been so so till now, not sure if this is the software issue or hardware. One improvement I noticed after Core 2.3.0 is that RSSI is 100, previously at the same place RSSI was around 72.

Frogmore42, 2018-08-17:
Like with most things there is a learning curve.  Tasmota's primary purpose is not the web UI.  That is there primarily for ease of configuration.  Once you have things set up you will likely never need to visit it again.  I have devices running Tasmota that I have not visited for months.

As far as value goes, the hardware is very inexpensive.  Even a few years ago equivalent hardware would be 2-10 times as much.  The software is an incredible value, even if you include the cost of your own time.  I know when I first was looking for solutions, I wanted something that just worked.  I found nothing like that.  But I did find Tasmota, which sounded like it might work if I put some effort into learning how to use it.  The documentation is better now, but when I started it was quite confusing.  It seemed to require installing a bunch of stuff just to try it out.  With some effort, I was able to figure out enough to get done what I wanted.  My first goal was automating the vent fan in my bathroom.  It now comes on automatically when I take a shower and stays on until the humidity returns to normal.  Alxea also knows how to turn the fan on and off.  That was on my list of projects for a decade, but doing it from scratch was not something I ever had enough time to do.  Sonoff, Tasmota, MQT, NodeRED, and some other bits made it pretty easy.

ecorman, 2018-08-18:
I uploaded last tasmota version (yesterday) I'm fully new then but I also observed web server stopped in some cases and in particular when I send a Websend command.
Is this a known issue ?

Geitde, 2018-08-20:
@lazers378 

> if this is the software issue or hardware.

It is not the hardware. I have 16 devices mounted inside the walls and they keep vanishing and being unreachable over time.

Frogmore42, 2018-08-21:
@geitde how do you know it is not the hardware?
This particular original issue was caused by using a known bad version of the core software.  The web server code in the core is known to not be 100% reliable.  Having said that, I have devices that I have had for many months that are still running the original software that I flashed months ago when I deployed them.  Since we have power failures every month or two, none have been continuously powered for that length, but other than a couple of failures very early all are still working fine.  All are currently running various 5.x versions.  The switch to the 2.4.x cores has been problematic and there were some other changes that made updating complicated for me.  So, I don't have personal experience with the 6.x versions.  As the saying goes, your mileage may vary.

Geitde, 2018-08-21:
I have running 16 devices and they randomly drop out and only cutting mains ins the fuse box works to get them back. My hardware is based on different brands and in use for 24/7.

You said yourself the code is not 100% reliable. Since you have power failures, you simply did not run into this.

Here they are in 24/7 use. I even had a sonoff touch, which completely locked up, not allowing to even switch via touch sensor anymore. The devices fail to respond to alexa stuff randomly, they fail to perform webserver stuff randomly and sometimes they stop working completely.

It would be the hardware, when I would have used one device. But having the same experience over 16 devices is unlikly a hardware issue.

If it is so hard to get this fixed, one should build in a scheduled reboot within the night. That would at least hide the issues.

arendst, 2018-08-21:
Build your own scheduled reboot using available rules.

Frogmore42, 2018-08-21:
Rules are proving to be very powerful. So that is one way to "solve" your issue.

I would instead take more data and analyze it for patterns.  For example, I have one device that is very problematic.  It is a NodeMCU.  In fact all of my NodeMCU devices are problematic.  I am still using this one because for the task it is doing it isn't that important that it is always up.  It varies in terms of how long it can stay up, sometimes only a few mins at other times it can be up for days before it reboots.  Since I had an ADC on the 3V3 line, I could tell it was not stable.  This caused an issue with the sensors I was using, so I added a another regulator just for the sensors.  It turned out this one was also defective, because when I selected a different one I got a really stable measurement (+/- 1mV over a long period).  Clearly, the instability was caused by hardware.

Another thing that happens at my site is the power goes out when a tree falls on a power line.  Since this happens with some frequency, I have an automatic standby generator.  This works great, but there is a delay.  This wasn't a major issue until I started using a POW.  The POW is dependent on the time being correct to keep daily totals.  It took a while, but I discovered that when there is a network but no NTP server, the code can get into a state where it will not recover the time, until the device is removed from power for about 30s or so, or reflash by serial (but that might be because it always takes me that long to set things up).  I traced this to the underlying core code, but it might even be in the SDK.  I am going to set up my own NTP server, since this is likely the best solution for my site.

Anyway, back on topic, if you want the auto reboot feature, you can have it with rules.  If you want to help make things better, start collecting more data to get to the real issue.  Perhaps it is running low on heap.  Perhaps the issue only occurs when the power is low, either line or from the regulator.  Perhaps it is when a certain level of traffic or errors or interference on the network occurs.  But, best to stop discussing this somewhat off topic stuff on this issue.

wongnam, 2018-08-23:
@Geitde be careful about the relay will flicker(in ms) If you create schedule to reboot within the night.

ascillato, 2018-08-23:
Hi,

@lazers378 

Please, Try to erase all flash and install last Tasmota with the Tasmota_mqtt library, and esplib v2.4.2

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-08-27 by ascillato2

lazers378, 2018-08-28:
Using Core 2.3.0 does seem to improve at least Wifi signals. However, web server issue is still there. Since the timer and openhab integration is working fine for last 10 days so I guess that is ok. 
Regarding flashing with esplib 2.4.2 and lwlp 1.4, I am not sure how to do that.