arendst/Tasmota #3500: Not getting assigned IP/chipld/ after connecting using smartconfig

amitavamighty, 2018-08-16:
I am developing an Android app for my IOT device using esp8266 flashed with sonoff tasmota. I am using smartconfig to push my SSID into esp8266. It's working perfectly. But I need either of chipid or the assigned IP address for further functionality, which I am not getting after successful connection in my android app.
Any help in this regard will be highly appreciated.
Thanks.

Jason2866, 2018-08-17:
You could do a mDNS query. Tasmota has a mDNS client built in.

amitavamighty, 2018-08-17:
Sorry. Didn't get it. To make further queries I need either the up of the device(i.e esp8266 controller) or the chip I'd. Can you please explain @Jason2866

Jason2866, 2018-08-17:
The ip address is assigned from the DHCP server (in most cases the internet router).
So the device has at the moment you do config Smartconfig has no IP of the target network.
So the only possibility is using the common know protocol mDNS to get in touch with the device.
The chip ID cant be read at config moment too.
Tasmota is designed to use mqtt. With mqtt there is no knowledge neccersary what IPs devices have.
Every function based on IP or webconsole (http) addresses is a nice to have enhancement

Jason2866, 2018-08-17:
YETI (android app) uses mDNS to find the devices and has the option to add the IP manually.

amitavamighty, 2018-08-17:
@jason may be I have placed my question wrongly. Actually I want to get the esp deviceid/ chipid along with the success message of smartconfig in my app. Please suggest how to do that.

Jason2866, 2018-08-17:
Sorry i dont know if this is possible.