arendst/Tasmota #3501: Dallas sensor error

Dylantje, 2018-08-16:
Dear..
When i try to activate dallas in the sketch i get some errors:
`xsns_05_ds18x20_legacy:39: error: redefinition of 'uint8_t ds18x20_address [8][8]'

 uint8_t ds18x20_address[DS18X20_MAX_SENSORS][8];

                                               ^

xsns_05_ds18x20:41: error: 'uint8_t ds18x20_address [8][8]' previously declared here

 uint8_t ds18x20_address[DS18X20_MAX_SENSORS][8];

         ^

xsns_05_ds18x20_legacy:40: error: redefinition of 'uint8_t ds18x20_index [8]'

 uint8_t ds18x20_index[DS18X20_MAX_SENSORS];

                                          ^

xsns_05_ds18x20:42: error: 'uint8_t ds18x20_index [8]' previously defined here

 uint8_t ds18x20_index[DS18X20_MAX_SENSORS] = { 0 };

         ^

xsns_05_ds18x20_legacy:41: error: redefinition of 'uint8_t ds18x20_sensors'

 uint8_t ds18x20_sensors = 0;

         ^

xsns_05_ds18x20:43: error: 'uint8_t ds18x20_sensors' previously defined here

 uint8_t ds18x20_sensors = 0;

         ^

xsns_05_ds18x20_legacy:42: error: redefinition of 'char ds18x20_types [9]'

 char ds18x20_types[9];

                     ^

xsns_05_ds18x20:45: error: 'char ds18x20_types [9]' previously declared here

 char ds18x20_types[9];

      ^

C:\Users\D.M. PC\Desktop\Tasmota 1\Sonoff-Tasmota-6.1.1\sonoff\xsns_05_ds18x20_legacy.ino: In function 'void Ds18x20Init()':

xsns_05_ds18x20_legacy:44: error: redefinition of 'void Ds18x20Init()'

 void Ds18x20Init()

      ^

xsns_05_ds18x20:244: error: 'void Ds18x20Init()' previously defined here

 void Ds18x20Init()

      ^

C:\Users\D.M. PC\Desktop\Tasmota 1\Sonoff-Tasmota-6.1.1\sonoff\xsns_05_ds18x20_legacy.ino: In function 'void Ds18x20Convert()':

xsns_05_ds18x20_legacy:94: error: redefinition of 'void Ds18x20Convert()'

 void Ds18x20Convert()

      ^

xsns_05_ds18x20:278: error: 'void Ds18x20Convert()' previously defined here

 void Ds18x20Convert()

      ^

C:\Users\D.M. PC\Desktop\Tasmota 1\Sonoff-Tasmota-6.1.1\sonoff\xsns_05_ds18x20_legacy.ino: In function 'boolean Ds18x20Read(uint8_t, float&)':

xsns_05_ds18x20_legacy:102: error: redefinition of 'boolean Ds18x20Read(uint8_t, float&)'

 boolean Ds18x20Read(uint8_t sensor, float &t)

         ^

xsns_05_ds18x20:286: error: 'boolean Ds18x20Read(uint8_t, float&)' previously defined here

 boolean Ds18x20Read(uint8_t sensor, float &t)

         ^

C:\Users\D.M. PC\Desktop\Tasmota 1\Sonoff-Tasmota-6.1.1\sonoff\xsns_05_ds18x20_legacy.ino: In function 'void Ds18x20Show(boolean)':

xsns_05_ds18x20_legacy:169: error: redefinition of 'void Ds18x20Show(boolean)'

 void Ds18x20Show(boolean json)

      ^

xsns_05_ds18x20:351: error: 'void Ds18x20Show(boolean)' previously defined here

 void Ds18x20Show(boolean json)

      ^

C:\Users\D.M. PC\Desktop\Tasmota 1\Sonoff-Tasmota-6.1.1\sonoff\xsns_05_ds18x20_legacy.ino: In function 'boolean Xsns05(byte)':

xsns_05_ds18x20_legacy:223: error: redefinition of 'boolean Xsns05(byte)'

 boolean Xsns05(byte function)

         ^

xsns_05_ds18x20:409: error: 'boolean Xsns05(byte)' previously defined here

 boolean Xsns05(byte function)
`
So i cannot upload the sketch..
When i set in the sketch the dallas option off.
I can use the Tasmota normal...
Hope someone can ex plane to me what i can do.

Thanks...

Frogmore42, 2018-08-17:
What change to the code did you make?
It looks like you have two conflicting \#defines defined.

Dylantje, 2018-08-17:
I did remove from the lines the // 
`// -- One wire sensors ----------------------------
                                                 // WARNING: Select none for default one DS18B20 sensor or enable one of the following two options for multiple sensors
#define USE_DS18x20                              // Optional for more than one DS18x20 sensors with id sort, single scan and read retry (+1k3 code)
#define USE_DS18x20_LEGACY                       // Optional for more than one DS18x20 sensors with dynamic scan using library OneWire (+1k5 code)
`

arendst, 2018-08-17:
So try to understand what the lines above mean in Dutch...

Verwijder van een van onderstaande defines de //. Dus niet van beide....

Dylantje, 2018-08-17:
Theo.... Thanks!!!
Onvoorstelbaar wat een mooie software!!!!!!
Sorry i am a noob...
Come from espeasy, only this is much better!!!
Only the setting learning as a bricklayer. pffff  :-)

Only i make great steps...

I have the problem thank!!!
close..

closed 2018-08-17 by Dylantje