arendst/Tasmota #3502: HX711

pronsta, 2018-08-17:
Hi, 

Is HX711 for the load cell supported by tasmota? 

thanks

Jason2866, 2018-08-17:
Not supported.

ascillato2, 2018-08-17:
Seems that your question has been answered. Closing issue. Please, ask to reopen if needed. Thanks

closed 2018-08-17 by ascillato2

arendst, 2018-10-08:
Stay tuned

mike2nl, 2018-10-20:
@arendst 
Saw it that the driver was now a tasmota one supported type. Well done...

- Question about the interface:
I saw a breakout board with one data and one clock connection and i mean that i have seen t
hat the driver uses two gpio"s? That's right?
One for GPIO_HX711_SCK and one for GPIO_HX711_DAT. 

Cool add-on to tasmota. Found something what i will build in the colder winter days:
https://www.instructables.com/id/Arduino-Load-Cell-Scale/

And the sensor is a 150KG with a precission of 0,02%, link: https://hackerstore.nl/Artikel/638.
The instrument amplifier will be this one so far i find nothing else: https://hackerstore.nl/Artikel/696

Ok the links are from a local dutch store but with picture everyone can find it.

arendst, 2018-10-20:
So what's the question?

arendst, 2018-10-20:
You provided all the answers already. The amp is the same as I use.

arendst, 2018-10-20:
[image]

[image]

[image]

JoschaMiddendorf, 2019-05-03:
Is ist possible to add multiple (3) HX711 to one NodeMcu module?

Jason2866, 2019-05-03:
No

frodo1990, 2019-12-03:
Hello, yesterday I tried to get the HX711 up and running. The calibrate works and a weight of 3Kg are also displayed. That works great. But as soon as I take the ESP from the power or reboot, then the display shows 0Kg. Even if the weight of 3Kg is still on the sensors.

I would like to build the sensors under a water softener, so that I can read the weight of the salt tablets and send it via MQTT.

How can I save the value after a reboot? I have the Tasmota version 7.1.1 on the ESP.

lalo-uy, 2019-12-03:
I have the same problem.
The module do not backup the calibration offset. It assume that power on is
with no load.
There is an option to backup the weight prior to reboot, but is useless in
a blackout.
I did not have time to make a PR.

darkdevil373, 2021-05-02:
Is there any new Information?

florinssk, 2022-10-08:
any new data about losing data after reboot ?

jaymemaurice, 2022-10-08:
Since this issue is closed, I doubt anyone is looking at it. Is it that the module itself tares to 0kg on power loss like most scales seem to do!? Maybe this is the wrong module for an application that always has weight on it… do we have a data sheet?