arendst/Tasmota #3503: Get esp deviceid or chipid as a success messsage for wifi smartconfig

amitavamighty, 2018-08-17:
Is it possible to get the esp deviceid or chipid along with the success message of smartconfig on my ESP8266 SmartConfig app or similar apps. Its necessary to identify which esp has successfully received the config if there are multiple esp's. Please suggest how to do that.

arendst, 2018-08-17:
Smartconfig is a concept of the espressif  SDK. Ask them about smartconfig options.

amitavamighty, 2018-08-17:
@arendst Can you tell me the file where I can check and edit the smartconfig responses/parameters in tasmota?
Thanks for the help.

arendst, 2018-08-17:
Using the correct tool look for ``smartconfig`` in file support.ino

ascillato2, 2018-08-17:
Seems that your question has been answered. Closing issue. Please, ask to reopen if needed. Thanks

closed 2018-08-17 by ascillato2