arendst/Tasmota #3505: Sonoff TH16 with Si7021 not showing sensor data

GerardRibas, 2018-08-17:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
17:52:49 MQT: stat/sonoffth16/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff"],"Topic":"sonoffth16","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:52:49 MQT: stat/sonoffth16/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:10","StartupUTC":"2018-08-17T16:50:39","Sleep":0,"BootCount":20,"SaveCount":41,"SaveAddress":"FB000"}}
17:52:49 MQT: stat/sonoffth16/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-07-31T17:59:09","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:52:49 MQT: stat/sonoffth16/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXX",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
17:52:49 MQT: stat/sonoffth16/STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","03B6179E","00000000"]}}
17:52:49 MQT: stat/sonoffth16/STATUS5 = {"StatusNET":{"Hostname":"sonoffth16-0279","IPAddress":"XXXXXX","Gateway":"XXXXXX","Subnetmask":"255.255.255.0","DNSServer":"XXXX","Mac":"68:C6:3A:CE:21:17","Webserver":2,"WifiConfig":2}}
17:52:49 MQT: stat/sonoffth16/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CE2117","MqttUser":"tasmota","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:52:49 MQT: stat/sonoffth16/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 17 16:52:49 2018","Local":"Fri Aug 17 17:52:49 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:45","Sunset":"20:02"}}
17:52:50 MQT: stat/sonoffth16/STATUS10 = {"StatusSNS":{"Time":"2018-08-17T17:52:49"}}
17:52:50 MQT: stat/sonoffth16/STATUS11 = {"StatusSTS":{"Time":"2018-08-17T17:52:50","Uptime":"0T00:02:11","Vcc":3.192,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXX","RSSI":74,"APMac":"94:A7:B7:E1:9B:DC"}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi,

I just flashed 6.1.1 in Sonoff TH16 with Sensor si7021.

I configured the module as `04 Sonoff TH` and I selected in GPIO14 Sensor `03 SI7021` I restarted the Sonoff multiple times but I still don't get any data from the sensor. 

Is there anything that am I missing?

Thanks.

Frogmore42, 2018-08-18:
Do you have an iTead si7021?  or is it a generic one?
Is it plugged all the way in?
I have a TH16 that has a defective jack.  I had to replace it to get any sensor to work.

ascillato, 2018-08-23:
@GerardRibas

Hi,

Have you managed to solve your issue?

closed 2018-08-25 by ascillato2

GerardRibas, 2018-08-30:
Hi guys,

Sorry for the delay, all seems working as expected. I replaced the jack as @Frogmore42 proposed so happy to close it.

Thanks!