arendst/Tasmota #3506: Sonoff POW not responding after power failure

GileanEU, 2018-08-17:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Sonoff POW and Sonoff TH-16
- [x ] Provide the output of command ``status 0`` : not possible
```
STATUS 0 OUTPUT HERE
```

I am not if this is the right place to ask, but here's my problem.
I have two setups side by side, like this: a power plug extension cord -> Sonoff POW -> Sonoff TH16 -> power plug.
This way I have been able to have both power consumption and temperature readings in my cellar that is under my outdoor shed. Both setups have Tasmota installed and they have been working just fine. Until 2 days ago, when I had 7 minute long power failure.

After power came back, one of those POW-TH16 -pairs came back, but other pair didn't. I noticed later from Home Assistant that this other pair was not reachable, and history showed that it had been so since that power failure. That pair does not get IP address either, so I cannot do anything via UI. I unplugged the extension cord and plugged it back, and then Sonoff POW blue led flashed once. I opened the cover (I have IP65 box with transparent cover for each Sonoff) and button did not do anything. Not multiple presses or even that holding down the button for 40 seconds. The only response I have from that Sonoff is when I unplug/replug the whole thing, and then I get that short blue blink once.

What can I do? I have these systems quite permanently installed on my shed's wall so I really wouldn't want to take them off and re-code everything, but how can I restore life to that Sonoff without tearing my installations apart? Also, if this happens again when the next power failure happens, I probably will have to do it again? However, there has been power failures before without problems, and even this time, the other setup right next to this one is working fine still.

GileanEU, 2018-08-18:
Thanks for reply, I wouldn't have believed they are so sensitive to power failures, but then again, they are not very expensive either so I guess that's what one can expect for that price :) Unfortunately this setup is in my "stairway room" to give power to cellar radiator and wires (both power line and TH16 sensor) go through the wall to cellar because wifi signal is too weak if I put them all in cellar, and that's why there are a bit difficult installations there. Including that IP65 cover box because it can get even -20C in that room where they are during winter. The other installation right next to these ones give power to water pump in that stairway room and that is easier to replace but of course it was the more difficult one that got broken :) Luckily it's not winter yet so my cellar does not need heating and I have time to fix this.

Although if POW is dead, TH16 behind it might still be working, it just cannot work without power from POW. I am not 100% sure if it's broken though because I remember Sonoff (or Tasmota?) went into same state once (only blue led blinked once) but I got it working by reflashing. But I will order spare ones anyway, because I have to remove my setup and bring it inside for reprogramming anyway.

Frogmore42, 2018-08-18:
Yes, it is good to have spares.  I usually use some type of connection that is easy to remove, so I don't have to unscrew small screws in difficult locations.  You might be able to recover the POW that is dead, but how much would you trust it in that location.  If you are good at troubleshooting, you might be able to repair it.

ascillato2, 2018-08-21:
Hi,

If you think that your question has been answered, please close this issue. Thanks.

ascillato2, 2018-08-25:
Hi,

Any news?

GileanEU, 2018-08-25:
Unfortunately no, except that I cannot see Sonoff in my wifi list anymore. I have been very busy with other tasks and now I am leaving for a holiday for 5 days, so cannot fix this before I come back :/

GileanEU, 2018-09-01:
All right, now I had time to investigate this more. I pushed the button, and only once I briefly saw sonoff on my wlan list, but it disappeared after 2 seconds. After that, it never came visible again. So, I pulled it off from my shed wall, and tried reflashing. That went without problems, BUT pushing button four times did not help: I couldn't find that on my wlan list. Apparently broken.

Then I flashed my new spare Sonoff POW, and tried the same: that one did appear on my wlan list. And after tweaking with settings, things worked again. Also, when I plugged the wires back in on my shed wall, TH-16 behind this POW also came back online! So it was only this one POW that broke because of this power failure.

ascillato2, 2018-09-03:
Ok, So closing issue as it was hardware defect. Thanks for share your experience.

closed 2018-09-03 by ascillato2

Otterjasie, 2022-06-04:
I got this same issue today on all my Sonoff Basic devices (5 of them) flashed with Tasmota 8.5.1.  This was caused by the following chain of events:   the refrigerator compressor developed an earth leakage that tripped the power to the house. To isolate the problematic circuit I impatiently tried to reset the earth leakage unit repeatedly while isolating different sub-circuits to isolate the fault. In hindsight that obviously caused numerous short power-ups to all the Sonoffs somehow damaging/clearing their setups. Luckily there were NO HARDWARE FAILURES of the Sonoffs.  I downloaded their different setups from my Dropbox backups and everything is fine again.  Would a longer delay after power-up in Tasmota SW prevent damage to the config memory to make these devices foolproof in such a situation?  A longer delay will allow more time for the power to stabilise.

pmknowles, 2022-06-04:
You probably triggered the Device Recovery process

Device Recovery - Tasmota<https://tasmota.github.io/docs/Device-Recovery/>

Regards,

Phil

Otterjasie, 2022-06-04:
Thank you for your quick reply and information.   I most certainly exceeded the 7 quick simultaneous power-ups, triggering "Device-Recovery" without intending to do it.    This trap is too easy to step into during earth leakage debugging , more so for people not aware of all these intricacies.