arendst/Tasmota #3507: Sonoff 4CH pro as Fan Coil thermostat

Lumpusz, 2018-08-17:
Hi,

I was wondering if a sonoff 4CH pro could handle the job of a fan coil thermostat, with an attached thermal sensor. I put together some requirements that I would consider the bare minimum:

- 3 speed fan control
- 2 pipe vavle control
- Summer/Winter switch
- Manual mode, if Wifi is down, but electricity is still on
- Recovery after power outage
- Integration with Home Assistant

Do you know any limitation either HW or SW that would prevent me hacking together this? If this uint is not up to the task, can you recommend anything else? I was also looking at KNX systems/thermostats, but they are not cheap.

Thanks.

Disclaimer:
I have only met tasmota/sonoff and home automation a week ago, so I am pretty new to this stuff, but I think this is amazing.

Frogmore42, 2018-08-17:
That is likely more I/O than a single esp8266 can handle.  There are ways to do it, but it won't be easy.

Frogmore42, 2018-08-18:
You didn't specify your background.  Depending on that, this might be easy or actually incredibly hard.  Also, depending on your requirements (what you have listed are not requirements, but rather features), and constraints, both functional and non-functional, it is possible that you could do it with a pro.  The Pro has 4 relays, but they are SPDT so you could get a little more flexibility.  But, my guess is that you probably need more than four relays.

You also need to think about how you handle failure and how bad would things get if your system stops responding, eg would the coil get too hot or cold and be a danger.  Would energizing all the relays be bad, what about none?

You probably want to do this thinking before you wire it all together and discover it really is a problem and now some component is destroyed.

ascillato2, 2018-08-21:
Hi,

If you think that your question has been answered, please close this issue. Thanks.

closed 2018-08-22 by ascillato2