arendst/Tasmota #3508: dynamic memory allocation for MQTT in TLS mode & local web service

orcema, 2018-08-17:
Combined usage of mqtt service in TLS mode and the local web service is impossible due to insufficient memory,

Why don't we try to use dynamic memory allocation for the web server as well for the mqtt service.
Thus if the mqtt service in TLS mode encounter problems connecting to server it will disconnect, free memory and launch the the local web server to allow the user to check for problems. The user can then proceed to updates and launch a new mqtt connection attempt.
New mqtt connection attempt will stop the local web server in order to free memory and lanuch the mqtt service. This process can repeat until successful connection to mqtt server.
Would this be a potential solution to insufficient memory ?

Frogmore42, 2018-08-17:
You don't need the web server.  If you need/want TLS turn off the web server.  Use the serial connection to verify everything is okay.  If you are running a local MQTT broker you don't really need TLS.  If you are running a broker on the open internet, TLS is probably a good idea, but it probably also won't work with the limited RAM in the esp8266.  If you want all these expensive security features, you need a more capable controller, which will likely cost a lot more money.

orcema, 2018-08-18:
Indeed you only need TLS if the broker is running on the open internet. I thought it would be i nice feature allowing new user to make use of the mqtt services without the need to implement a local mqtt broker.
I already tried with some code snippets to do this dynamic memory allocation and it worked well.
So if there is an interest i could try to implement this feature by a pull-request.

Frogmore42, 2018-08-18:
Many people who have tried to use a broker on the internet have had issues.  Maybe those issues were related to not really having enough memory.  I found it reasonably easy to set up mosquito, but I am sure someone else might find it nice to use a remote broker.

ascillato2, 2018-08-27:
> I already tried with some code snippets to do this dynamic memory allocation and it worked well.

Can you share those?

ascillato2, 2018-09-04:
Closing as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-09-04 by ascillato2