arendst/Tasmota #3510: Possible to read switchstate after manually setting a gpio to high?

cobradevil, 2018-08-18:
Hi all,

I'm trying to get a soil moisture sensor working using tasmota. Using the lm392 with a digital output this is no problem except that there is always current on the sensor which erodes very quickly.
What i would like to do is set the gpio to on wait a second and then read the state and then turn it off again so the sensor does not erode that quick and to save some power.
I'm not sure if this is possible and after searching the issue list and forums I could not find an answer, any directions or thoughts on this would be nice.

Best regards,
William

ascillato, 2018-08-18:
Hi,

The LM392 is a _LOW-POWER OPERATIONAL AMPLIFIER AND VOLTAGE COMPARATOR_ that does not erode with continuous use. Besides, It should be better to use a complete moisture sensor module (very inexpensive) that will provide you a % of soil moisture (you can attach it to the ADC input of your Sonoff and there is no need to configure anything - just compile Tasmota with ADC input ON). I'm using that approach and works really good.

Anyway, If  you want to power your LM392 off, may be you will need to use an extra GPIO as the voltage source. You can attach a transistor to make it power the lm392. Then you can set a RULETIMER to power that GPIO on, and to check the SWITCH state, and then power it off. It is kind of complex (because it is very specific) but you can make it using rules.

Hope this helps.

Frogmore42, 2018-08-18:
@cobradevil is likely concerned about galvanic corrosion on the pcd fingers, which is real and likely.  While turning it off will probably help, those sensors are not long lived or very accurate or very reliable.  They are, however, very cheap.  A really good sensor for this purpose is quite expensive.  I haven't tried it yet, but this is the one I would use.
https://www.vegetronix.com/Products/VH400/