arendst/Tasmota #3513: Sonoff T1, Serial Programming fails.

FabrizioRomanoGenovese, 2018-08-18:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff T1, Raspberry Pi3B+
- [ ] Provide the output of command ``status 0`` :

I have a Sonoff T1 UK that unfortunately comes with firmware 1.8, so it cannot be flashed with SonOTA. Having to flash it via serial, I wanted to use my Raspberry Pi3B+ as a serial programmer. Flashing with a usb programmer is unfeasible for me because of the operating system I am using on my laptop, so please do not propose this as a solution.

I used the following guide: https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi

- I soldered the wires on the Sonoff, and I connected them to the pi:
3.3V <-> Pin 1
RX <-> Pin 8
TX <-> Pin 10
GND -> Pin 39
- After following the instructions on the guide, I am able to detect /dev/sstyS0 on the pi;
- I installed `esptool`;
- After following the instructions on the guide, I am able to put the T1 into flashing mode by holding button 1, then pressing button 4, then releasing button 1 after 2-3 secs. The wifi led is off and the button led is lit and dimmed;

Now, when I try to erase the flash memory with `https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi` I get `Connecting...`, which after some seconds gives me `A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header`. 

What do I do?

I tried:
- Inverting TX and RX wires;
- Connecting 3V3 and GND to an external, professional current generator for electronics, stable at 3.3V;
- Keeping button 1 down while flashing;

Nothing of the above seems to work.

Frogmore42, 2018-08-18:
That error generally indicates you don't have it connected correctly.  You should try a terminal emulator and have use what you think is the correct serial port.  Every time the esp8266 boots it sends out information on its serial port.  If you can't see this on the terminal emulator, you don't have it connected correctly or you don't have the correct port or you don't have it configured correctly.  Since there are only three wires necessary to receive data, it shouldn't be too hard.  But, many people have struggled to do it the first time.

FabrizioRomanoGenovese, 2018-08-19:
>  You should try a terminal emulator and have use what you think is the correct serial port.

What do you mean? I am using esptool via terminal, but I suppose what you are proposing is different.

Frogmore42, 2018-08-19:
To verify your connections are correct use a terminal emulator to listen to the serial line.  When the esp8266 starts it always outputs something.  If you see nothing, you have something wrong.  You can also try connecting the RPi's Tx and Rx together to see that you have the correct ones.

What terminal are you using to run esptool?
The RPi only has a single serial port as I recall.  The configuration required removes the use of the serial port from the OS, so it is available to an application, like putty.

ascillato, 2018-08-23:
Hi,

Have you managed to solve your issue?

FabrizioRomanoGenovese, 2018-08-24:
Yes, sorry.

At some point it decided to work without me doing anything different. I cannot give any explanation of what's changed. :/

closed 2018-08-24 by FabrizioRomanoGenovese