arendst/Tasmota #3514: Unable to compile. 'RtcSettings' was not declared in this scope

andreasloe, 2018-08-18:
- [X] Searched the problem in issues, found it with a tipp that did not help: https://github.com/arendst/Sonoff-Tasmota/issues/1830
- [X] Searched the problem in the wiki
- [X] Searched the problem in the forum, nothing found
- [X] Searched the problem in the discussion group, nothing found
- [X] Hardware used: Mac OS 10.13.6, arduino 1.8.5 
- [ ] Provide the output of command ``status 0`` :
I am talking about the source code here.

I cannot compile with a fresh installation of arduino and Tasmota 6.1.1 20180714. The error message is 
```
/Users/andies/Documents/Arduino/Sonoff/settings.ino: In function 'uint32_t GetRtcSettingsCrc()':
settings:75: error: 'RtcSettings' was not declared in this scope
   uint8_t *bytes = (uint8_t*)&RtcSettings;
                               ^
settings:77: error: 'RTCMEM' was not declared in this scope
   for (uint16_t i = 0; i < sizeof(RTCMEM); i++) {
```
The full error message is available here: https://pastebin.com/RnmAGdfK

Frogmore42, 2018-08-19:
Most people here don't have a Mac.  You already found the other issue where someone else was unable to get it working with Arduino on a Mac.  It works with Arduino on Windows, based on other people's experience.

If you are not experienced in software development, it might be best to use one of the pre-built binaries.  Or if you are up for a little challenge, use PlatformIO.  I have not heard of issues with the Mac with it.  It might seem a little intimidating at the beginning, but it is a really powerful environment.

andreasloe, 2018-08-19:
But isn't this a _pure software issue_? (Will try PlatformIo nevertheless.)

Frogmore42, 2018-08-19:
The Arduino environment is not really great for something as large and complex as Tasmota.  The complier is either getting confused or you really are missing something.  Since the code compiles correctly in other environments, it must not always be wrong.  Since it isn't working for you on a Mac and it has also not worked okay for other people on a Mac, it is possible the Mac requires some additional configuration to work right.  If you are up for the challenge, since you have a Mac and can reproduce the issue, perhaps you can figure out what is missing or misconfigured.

If your primary goal is to just use Tasmota, use one of the precompiled binaries and save yourself the hassle of compiling.  I believe the binaries are in a good state currently and you can configure everything through the serial interface, so there is no real need to compile.

andreasloe, 2018-08-19:
Just built it successfully on PlatformIO,
`======================== [SUCCESS] Took 568.05 seconds =======================`
Those, who come across this page: It is not a software problem, but an Mac-arduino IDE-issue.

Thanks for the help.

closed 2018-08-19 by andreasloe