arendst/Tasmota #3515: POW R2 reports 0W until rebooted

manutenfruits, 2018-08-18:
I have a few Sonoff POW 2 laying around the house, but the one connected to my TV seems to have a bug. I calibrated it and it reports the consumption just fine, but after a few hours (not sure what triggers it) it just starts reporting Power 0, even when the TV is on, and it records current. Check out a sample:

`17:23:16 MQT: tele/sonoff-tv/SENSOR = {"Time":"2018-08-18T17:23:16","ENERGY":{"Total":0.783,"Yesterday":0.000,"Today":0.130,"Period":0,"Power":0,"Factor":0.00,"Voltage":124,"Current":0.521}}`

Power is 0, but it should be voltage * current = 124 * 0.521 = 64.6 W. A reboot fixes it, but you never know for how long. I've tried reflashing.

Thanks!

manutenfruits, 2018-08-20:
Thanks for the help! Do you have a link to the issue? I can't find the right keywords to find it.

ascillato2, 2018-08-21:
Hi,

If you think that your question has been answered, please close this issue. Thanks.

manutenfruits, 2018-08-21:
Thanks! I'll flash that version and try again, I'll reopen the issue if it doesn't work.

closed 2018-08-21 by manutenfruits