arendst/Tasmota #3521: SDM120 when reporting negative values will send wrong info to Domoticz

gajotnt, 2018-08-19:
**Describe the bug**

_A clear and concise description of what the bug is._

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues and in the wiki
- [ ] Hardware used : SDM120
- [ ] Provide the output of command ``status 0`` :

10:58:55 MQT: domoticz/in = {"idx":48,"nvalue":0,"svalue":"**65530;4.196**","Battery":84,"RSSI":5}
10:58:55 MQT: tele/powcasa/SENSOR = {"Time":"2018-08-19T10:58:55","ENERGY":{"Total":4.196,**"ActivePower":-6**,"ApparentPower":183,"ReactivePower":-183,"Frequency":50.04,"Factor":-0.03,"Voltage":240,"Current":0.727}}

gajotnt, 2018-08-19:
My guess its because of the char data being 16? 

> The range of the char type is 0 to 2^16 - 1 (0 to 65535).

```
void DomoticzSensorPowerEnergy(uint16_t power, char *energy)
{
  char data[16];
  snprintf_P(data, sizeof(data), PSTR("%d;%s"), power, energy);
  DomoticzSensor(DZ_POWER_ENERGY, data);
}
```

arendst, 2018-08-19:
As a workaround could you try this by changing line 259 in file xsns_23_sdm120.ino from
```
      DomoticzSensorPowerEnergy((**uint16_t**)sdm120_active_power, energy_total);
```
Into
```
      DomoticzSensorPowerEnergy((**int16_t**)sdm120_active_power, energy_total);
```

And change line 349 in file xdrv_07_domoticz.ino from
```
void DomoticzSensorPowerEnergy(**uint16_t** power, char *energy)
```
in to
```
void DomoticzSensorPowerEnergy(**int16_t** power, char *energy)
```

and report back the result

gajotnt, 2018-08-19:
Worked like a charm :D

closed 2018-08-19 by gajotnt