arendst/Tasmota #3523: 3V / 4.5V chain of lights to nodemcu (wiring)?

prankousky, 2018-08-19:
Hi all,

I run Sonoff-Tasmota on all my supported sonoff devices as well as on some nodeMCU devices (Wemos alternative).

I am pretty sure that what I need it to do is possible, however, I am not sure how to wire this; should be very simple as there are only POWER and GROUND available...

My chain of lights ( / fairy lights??) are currently hooked up to a 3x 1.5V battery pack with an on/off switch. I have experimented with these lights before and they seem to actually run on anything in between 3V and 5V (while only 4.5V are supported according to product info).

How would I connect them to my nodemcu board running tasmota? GROUND to GND seems easy enough for me (obviously not a pro, just beginner/hobbyist). However, I am not sure what to connect the POWER cable to. If I use VIN or 3V, the lights will be on constantly, right? And I would not be able to switch them.

My goal is to turn the lights on and off via mqtt. 

Thanks in advance for your help 👍

Frogmore42, 2018-08-19:
You will need a relay.  There are lot's of tutorials on the internet for connecting a relay to a nodemcu.  Find one and follow the instructions.  This is not an issue with Tasmota.

Here is an example:
https://www.instructables.com/id/Interface-Relay-Module-With-NodeMCU/

Jason2866, 2018-08-19:
The IO pins from the NodeMCU cannot power your lights. You need a decent power supply or use your battery pack. Easiest and safest way is to use a relais driver board.
For WemosMini you can buy this ready assembled.
https://www.aliexpress.com/wholesale?catId=0&initiative_id=AS_20180819080645&SearchText=wemos+relay+shield

ascillato, 2018-08-19:
Hi,

If you think your question has been answered, please close this issue. Thanks

roguestreak, 2018-08-20:
You can also instead use a suitable transistor / FET, which can be used as basically a solid-state relay; a controlled switch, where the control is isolated from the switched circuit.  Here's some basic stuff which may set you on the right path
https://learn.sparkfun.com/tutorials/transistors/applications-i-switches

Now, the thing you need here is HOW to switch it.  You need to use one of the spare GPIO pins (eg. D4), and have it setup in tasmota to be able to switch that High or Low; do some searching for how to control a relay with tasmota, and the exact same thing can be applied to using a transistor.

I'm guessing you have about 20 LEDs, that's about 400mA that needs to be switched, which is not much at all but generally too much to run directly through a microcontroller, hence the need for an intermediary switch that you control using the microcontroller; a relay, transistor etc.

closed 2018-08-20 by prankousky