arendst/Tasmota #3525: GPIO9,10 not exposed for Generic device

SupraJames, 2018-08-19:
For 'Generic' device, sonoff_template.h reserves GPIO pins 7-12 for 'flash control' use. however my Sonoff T1 units use GPIO 9 and 10 for their second and third buttons.

I've made local changes to my source and I've changed GPIO 9 and 10 to GPIO_USER and it's working fine.

Is this a difference between ESP8266 and ESP8285 devices?

ascillato, 2018-08-19:
> Is this a difference between ESP8266 and ESP8285 devices?

Yes, exactly

That GPIO pins are not going to be exposed due to will make more problems that it solves.

See explanations at #3220, #2783, #2522

Thanks

closed 2018-08-19 by ascillato2

SupraJames, 2018-10-18:
For anyone who might come across this... All Hail SetOption81 ... :)

ascillato, 2018-10-18:
SetOption51

SupraJames, 2018-10-19:
Hmm it’s definitely 81 in 6.2.1 it’s the last option in the second block of options in settings.h

ascillato, 2018-10-19:
Yes, but In last 6.2.1.17 is setoption51

arendst, 2018-10-19:
Correct. See #3709. Still undocumented though ;-)