arendst/Tasmota #3526: WiFi is not connecting because of underscore in SSID

hansnull, 2018-08-19:
My device (Obi 2291706, Tasmota v6.1.1) is not connecting to my WiFi network.

**To Reproduce**
Create a WiFi network with a SSID which contains an underscore, i.e. "wifi_nomap".

**Expected behavior**
Just connect to the WiFi network.

If I remove the underscore the device is connecting, but I do need the "_nomap" in my SSID to opt out the google location services (https://support.google.com/maps/answer/1725632).
Any help appreciated : )

curzon01, 2018-08-20:
Can't reproduce, my test AP name is "`<*_*>`":

```
09:31:54 CMD: AP
09:31:54 MQT: dev/stat/tasmota/esp02/RESULT = {"Ap":"1 (<*_*>)"}
09:32:51 CMD: Status 11
09:32:51 MQT: dev/stat/tasmota/esp02/STATUS11 = {"StatusSTS":{"Time":"2018-08-20T09:32:51","Uptime":"0T18:40:13","Vcc":3.012,"POWER1":"ON","POWER2":"ON","POWER3":"ON","POWER4":"ON","Wifi":{"AP":1,"SSId":"<*_*>","RSSI":100,"APMac":"9C:C7:A6:46:A8:A3"}}}
```

May be our AP has a problems with underscores, not Tasmota. Does your AP work with other devices (may be mobile) and same name.

hansnull, 2018-08-20:
Yes, I have several other devices connected. I will test your AP name, when I am home.

hansnull, 2018-08-20:
I got it working now, the problem was not the underscore... it was a number.

closed 2018-08-20 by hansnull