arendst/Tasmota #3530: iFan02 PowerOnState doesn't restore fan

rdbahm, 2018-08-20:
When the iFan02 is power cycled (the device itself loses power) while the fan is on and PowerOnState is set to 3 (others not tested, but does not affect PowerOnState 0), the fan powers on with FanSpeed set to the previous value, but there is no fan power output.

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : iFan02
- [x] Provide the output of command ``status 0`` :
```
04:45:38 CMD: status 0
04:45:38 MQT: stat/officefan/STATUS = {"Status":{"Module":44,"FriendlyName":["Office Light"],"Topic":"officefan","ButtonTopic":"0","Power":7,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
04:45:38 MQT: stat/officefan/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:02:30","StartupUTC":"2018-08-20T03:43:08","Sleep":0,"BootCount":30,"SaveCount":118,"SaveAddress":"F6000"}}
04:45:38 MQT: stat/officefan/STATUS2 = {"StatusFWR":{"Version":"6.1.1","BuildDateTime":"2018-08-18T12:29:53","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
04:45:38 MQT: stat/officefan/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TheActualSSIDIUse","ARealSSID"],"TelePeriod":300,"SetOption":["400A8009","55818000"]}}
04:45:38 MQT: stat/officefan/STATUS4 = {"StatusMEM":{"ProgramSize":544,"Free":456,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000001","23B6179E","00000000"]}}
04:45:38 MQT: stat/officefan/STATUS5 = {"StatusNET":{"Hostname":"officefan-1234","IPAddress":"192.168.1.2","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:0D:8E:00:00:00","Webserver":2,"WifiConfig":3}}
04:45:38 MQT: stat/officefan/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.1","MqttPort":1234,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_42CE3A","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
04:45:38 MQT: stat/officefan/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 20 03:45:38 2018","Local":"Mon Aug 20 04:45:38 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:50","Sunset":"19:57"}}
04:45:38 MQT: stat/officefan/STATUS10 = {"StatusSNS":{"Time":"2018-08-20T04:45:38"}}
04:45:38 MQT: stat/officefan/STATUS11 = {"StatusSTS":{"Time":"2018-08-20T04:45:38","Uptime":"0T00:02:30","Vcc":3.486,"POWER1":"ON","FanSpeed":2,"Wifi":{"AP":1,"SSId":"ARealSSID","RSSI":84,"APMac":"00:00:00:00:00:00"}}}
```

**To Reproduce**
1. Set the fan speed on an iFan02 to a non-zero value.
2. Remove power from the iFan02.
3. Reconnect power to the iFan02.
4. The fan does not resume spinning - issue the command "FanSpeed" to see that the value matches that which was set in step 1.

**Expected behavior**
It is expected that the fanspeed in the firmware status should match the outputted fan speed.

**Additional context**
I have worked around this by setting PowerOnState to 0.

arendst, 2018-08-20:
Related to #3412.

The non-esp8266 microcontroller controls the power for both the light and the fan at power restore without sending it's status to the esp8266. 

At power on Tasmota tries to control the relays as defined by poweronstate but the microcontroller overrules this, probably because it's a bit slower than the esp8266 so it has final saying in which relay will be turned on.

Interesting to see that a restart of the iFan02 really does perform the poweronstate so this confirms my suspicion that the microcontroller is just a bit slower.

As in #3412 you currently need to use a rule to get the fans in the state you want. I'll investigate what needs to be done to send a single command using the current fanspeed set.

arendst, 2018-08-20:
Investigating. It seems it's different than I thought. Hold on.

arendst, 2018-08-20:
As the microcontroller controlling the relays only accepts commands after 4 seconds from power on you'll notice a delay in restoring the fan speed (and light).

Can't find a better solution.

closed 2018-08-20 by arendst

rishabmehta7, 2019-10-19:
Hi,

So now does tasmota on iFan02 have a retain state for light and fans speed?
Also do we see the speed control on Google assistant?
I am new to tasmota and still on factory firmware so wondering if I should flash tasmota and setup HA