arendst/Tasmota #3531: Provisioning (OTA auto-config)

udo1toni, 2018-08-20:
Would be nice to have a feature for autoprovisioning or at least manual provisioning. 

As the configuration backup is **not** human readable, it's not possible to create configuration for many similar devices and load the configuration after flashing.

One could create human readable configuration files, named `tasmota-mac.cfg` for example, with all options, values, names and so on, this files could be loaded via OTA, so if configured one firmware with OTA and a valid SSID/Password, Tasmota would load the firmware and afterwards the complete configuration, depending on its MAC Address.

It should be possible to im/export the settings in the same format as well.

wongnam, 2018-08-20:
i suggest using backlog to push parameter value to TASMOTA devices base on its fallback topic.

udo1toni, 2018-08-20:
Well, at least it is an option to push most of the configuration at once to a device, but there is a big difference between push and pull ;) i.e. I will have to build a script which uses backlog, will have to take care of fallback name... where a mechanism in tasmota itself would be: send a http get request with own mac address as payload to an address which is already configured.

Please think of VoIP deployment. I have 4 snom telephones at home... not that much, but it's nice to get telephone book and other stuff automatically when restarting the telephone. Even my wife and my parents are able to restart the telephone, and this is something tasmota could also do automatically after power failure - just take a look at the server and if there was something changed, load the new configuration.

arendst, 2018-08-20:
All nice and well but this human readable configuration will have to be translated to a binary configuration to be used by Tasmota. Try to imagine what amount of code is needed for this functionality which is only used at config download (and once made available they want upload too) time...

wongnam, 2018-08-20:
I used this way for my tasmota devices initial configuration only. it's only push the parameter to the new tasmota devices. i don't know how to pull too.
(sorry, it's a dirty way that i've applied to my stuff.)

[image]

[image]

`//!!Important: A backlog feature is available allowing to execute several commands in sequence (maximum 30/request).
var FallbackTopic = "DVES_853010" ;     // A Device FallbackTopic
var cmd1 = "topic balconylight" ;       //MQTT Topic
var cmd2 = "FriendlyName1 Balcony Light" ;   // (Default)"FriendlyName2 Sonoff1" to set Sonoff Friendly Name 
var cmd3 = "FriendlyName2 Sonoff2";     // (Default)"FriendlyName2 Sonoff1" to set Sonoff Friendly Name 
var cmd4 = "FriendlyName3 Sonoff3";     // (Default)"FriendlyName3 Sonoff1" to set Sonoff Friendly Name // (Default)"FriendlyName2 Sonoff1" to set Sonoff Friendly Name 
var cmd5 = "FriendlyName4 Sonoff4";     // (Default)"FriendlyName4 Sonoff1" to set Sonoff Friendly Name 
var cmd6 = "Module 10" ;        // Module type <index> eg: 1 (Sonoff Basic); 4 (Sonoff TH); 39 (Sonoff Dual R2); 10 (Sonoff Touch); 25 (Sonoff Bridge)
var cmd7 = "SetOption3 1" ;     // (default) "SetOption3 1"; MQTT support: 1=on; 0=off 
var cmd8 = "SetOption1 0" ;     // (default) "SetOption1 0"; Define button multi press mode. "SetOption1 0" Allow all button actions; Allow only single, 1 = double and hold press button actions
var cmd9 = "SetOption11 0" ;    // (default) "SetOption11 0" Legacy button single and double press; "SetOption11 1" send MQTT msg: cmnd/<non-topic>/POWER TOGGLE
var cmd10 = "ButtonTopic 0" ;   // (default) "ButtonTopic 0" Disable use of MQTT button topic; "ButtonTopic 1" Set MQTT button topic to Topic; "ButtonTopic <topic>" Set MQTT button topic (32 chars max)
var cmd11 = "PulseTime 0" ;     // (Default=0) Disable use of PulseTime<x> for relay<x> x0.1s
var cmd12 = "TelePeriod 60" ;  // (Default) "TelePeriod 300" Set telemetry period in seconds
var cmd13 = "SwitchMode 0";     // (default) "SwitchMode 0" Set switch mode to TOGGLE; 0=TOGGLE; 1=FOLLOW ; 3=PUSHBUTTON ; SwitchMode<x>
var cmd14 = "SerialLog 0" ;     // (default) "SerialLog 1" ; 0 or "off" = Disable Hardware Serial Bridge and disable serial logging
var cmd15 = "rule" ;            // set rules here. eg "rule on power1#Stat=1 do power2 on endon"
var cmd16 = "rule off" ;        // (default) "rule off" ; "rule on" to on the rules
var cmd17 = "GPIO1 0" ;         // Default "GPIO" Show current GPIO usage for current module; Select sensor to be connected to <pin> eg: 0 =none; 9 (Switch1); 2 (AM2301); 3 (SI7021); 4 (DS18x20)
var cmd18 = "GPIO3 0" ;         // Default "GPIO" Show current GPIO usage for current module; Select sensor to be connected to <pin> eg: 0 =none; 9 (Switch1); 2 (AM2301); 3 (SI7021); 4 (DS18x20)
var cmd19 = "GPIO" ;         // Default "GPIO" Show current GPIO usage for current module; Select sensor to be connected to <pin> eg: 0 =none; 9 (Switch1); 2 (AM2301); 3 (SI7021); 4 (DS18x20)
var cmd20 = "GPIO" ;            // Default "GPIO" Show current GPIO usage for current module; Select sensor to be connected to <pin> eg: 0 =none; 9 (Switch1); 2 (AM2301); 3 (SI7021); 4 (DS18x20)
var cmd21 = "GPIO" ;            // Default "GPIO" Show current GPIO usage for current module; Select sensor to be connected to <pin> eg: 0 =none; 9 (Switch1); 2 (AM2301); 3 (SI7021); 4 (DS18x20)
var cmd22 = "GPIO" ;            // Default "GPIO" Show current GPIO usage for current module; Select sensor to be connected to <pin> eg: 0 =none; 9 (Switch1); 2 (AM2301); 3 (SI7021); 4 (DS18x20)
var cmd23 = "WifiConfig 5" ; // (default=2) Start wifi manager (web server at 192.168.4.1) and set as current config tool. WifiConfig=5 Disable wifi config but retry same AP without restart and flash writes ;
var cmd24 = "Emulation 1" ;     // Emulation default 0=none; 1=WEMO; 2=HUE_Bridge
var cmd25 = "Timezone 7" ;      // Set local timezone 

if (msg.payload == FallbackTopic )
var msg1 = {payload : cmd1+";"+" "+cmd2+";"+" "+cmd3+";"+" "+cmd4+";"+" "+cmd5+";"+" "+cmd6+";"+" "+cmd7+";"+" "+cmd8+";"+" "+cmd9+";"+" "+cmd10+";"+" "+cmd11+";"+" "+cmd12+";"+" "+cmd13+";"+" "+cmd14+";"+" "+cmd15+";"+" "+cmd16+";"+" "+cmd17+";"+" "+cmd18+";"+" "+cmd19+";"+" "+cmd20+";"+" "+cmd21+";"+" "+cmd22+";"+" "+cmd23+";"+" "+cmd24+";"+" "+cmd25, topic:"cmnd/"+FallbackTopic+"/Backlog"};
    else msg1=null;

return [msg1];`

ascillato2, 2018-08-21:
Hi,

If you think your question has been answered, please close this issue. Thanks.

udo1toni, 2018-08-21:
@arendst Yes, fair point. 
Is there an easy way to decipher the binary config backup? Then it would be possible to build a tool for configuration handling, i.e. setting all parameters, automatically build the backup files, do an automatic restore through mqtt or command interface.

arendst, 2018-08-21:
Take a look at file settings.h which contains the Settins definition with memory location. If debugging is enabled and compiled you can even dump the config unencrypted...

ascillato2, 2018-08-28:
Hi,

Similar issue: #3181

closed 2018-08-28 by ascillato2