arendst/Tasmota #3532: Present RSSI value as a negative value

xbmcnut, 2018-08-20:
Hi there,

Running 6.1.0 and have brought the RSSI into Home Assistant but the value reports as a positive rather than the correct negative value.

`{"Time":"2018-08-20T11:56:32","Uptime":"0T14:34:18","Vcc":3.176,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","RSSI":70,`

roguestreak, 2018-08-20:
Looks like the "RSSI" in Tasmota isn't actually RSSI but "WiFi signal strength in percent"
Maybe do a rough conversion in HA using real_RSSI = (tasmota_RSSI / 2) - 100

ascillato2, 2018-08-21:
Hi,

@roguestreak is right. Tasmota is showing the WiFi signal strength in percent and not dBi.

If you think your question has been answered, please, close this issue. Thanks.

xbmcnut, 2018-08-22:
Thanks. Does that mean I need two sensors? One to define the RSSI and another to define the template required to invert the number?

ascillato, 2018-08-22:
> Does that mean I need two sensors? One to define the RSSI and another to define the template required to invert the number?

No, If you want only the signal strengh in dB just apply the formula in HomeAssistant when reading the JSON from Tasmota. Examples for that are in the wiki.

ascillato, 2018-08-23:
Hi,

If you think your question has been answered, please, close this issue. Thanks.

closed 2018-08-24 by xbmcnut