arendst/Tasmota #3533: MQT TasmotaMQTT rc 8

meingraham, 2018-08-20:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)

Referring to #2930

I am getting MQTT failed connections on one light in particular. The RSSI values are consistently low (30-50). However, the Wi-Fi appears to remain connected throughout. It is only the MQTT connection that drops (I set WEBLOG 4 on the device).

I thought that perhaps I had a questionable Sonoff Basic. I swapped it out for another and the behavior is similarly poor. It appears that it may just be the location of the switch (behind a dresser in the bedroom) rather than the switch itself. Nevertheless, I'm trying to troubleshoot it to see if I can improve the condition.

The failed connections come in spurts and there does not appear to be a pattern. I thought that perhaps my SLEEP setting was too high so I have disabled SLEEP altogether on this switch. This does not seem to have affected the situation.

In issue #2930, @arendst suggested that the TCP_RECONNECT_DISCONNECTING (rc 8) might be caused by a duplicate "Client ID". I'm not sure what "client" this refers to.

I have verified that each TASMOTA device has a unique MQTT topic.

I did a network scan and, as expected, every device on the network has a unique MAC address (thus the MQTT Client and Fallback Topic ought to be unique). I even verified that the three least significant octets of all MAC addresses are not duplicated (neither for the Sonoff devices nor for any device on the network).

The restart reason below states "Exception". Other than me setting the SLEEP parameter and the device restarting, I do not believe there were any other unexpected restarts. The time when I reset SLEEP the last time is the last restart of the device.

> 08:08:47 CMD: status 0
> 08:08:47 SRC: WebConsole from 192.168.1.xxx
> 08:08:47 RSL: Received Topic /status, Data Size 1, Data 0
> 08:08:47 RSL: Group 0, Index 1, Command STATUS, Data 0
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS = {"Status":{"Module":1,"FriendlyName":["MyFriendlyName"],"Topic":"MyMQTTTopic","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T08:08:15","StartupUTC":"2018-08-20T04:00:32","Sleep":0,"BootCount":21,"SaveCount":38,"SaveAddress":"F6000"}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS2 = {"StatusFWR":{"Version":"6.1.1b","BuildDateTime":"2018-08-02T15:54:49","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dd-wrt",""],"TelePeriod":300,"SetOption":["00008108","55818000"]}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS4 = {"StatusMEM":{"ProgramSize":424,"Free":576,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0F002B90","00000001","00000096","00000000"]}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS5 = {"StatusNET":{"Hostname":"MyMQTTTopic-DecMACID","IPAddress":"192.168.1.228","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:7A:40:2B","Webserver":2,"WifiConfig":2}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS6 = {"StatusMQT":{"MqttHost":"openhabianpi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_MAC3","MqttUser":"MQTTuser","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 20 12:08:47 2018","Local":"Mon Aug 20 08:08:47 2018","StartDST":"Sun Mar 11 02:00:00 2018","EndDST":"Sun Nov 04 02:00:00 2018","Timezone":99,"Sunrise":"06:50","Sunset":"20:09"}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS10 = {"StatusSNS":{"Time":"2018-08-20T08:08:47"}}
> 08:08:47 MQT: stat/MyMQTTTopic/STATUS11 = {"StatusSTS":{"Time":"2018-08-20T08:08:47","Uptime":"0T08:08:15","Vcc":3.170,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MySSID","RSSI":40,"APMac":"MyAPMAC"}}}

Any other areas I should investigate?

Thanks.

Mike

arendst, 2018-08-20:
And yet your MQTTClient is set as ``DVES_MAC3`` so likely to get duplicated...

meingraham, 2018-08-20:
@arendst <MAC3> is me removing the actual MAC information. I had enclosed "my" information in "< >" but the quoting removed everything w/in those so I had to remove them.

curzon01, 2018-08-21:
_MqttClient_ should be unique for each device. 

_MqttClient_ is an id where the MQTT connection are identified by the MQTT server. It is a technical value and there is normaly no reason to change the _MqttClient_ value other than default "DVES_%06X"

Reset _MqttClient_ to default using

`MqttClient DVES_%06X`

within Web Console

Hint: If you use code containing special chars like &lt; or &gt; mark the whole code witin the edit window and mark it with the **&lt;&gt;** (Insert code) from editor menu to prevent decoding possible special chars within your code text.

meingraham, 2018-08-21:
I wanted to verify that I have checked every possible item for duplicates to eliminate that possibility.

I have not found any clues as to why there would be TCP reconnect problems which cause the MQTT connectivity issues yet it appears from the weblog that the Wi-Fi connection is not dropping. If anybody has any suggestions on where to look, please point me in that direction.

Thanks for your help.

Mike

meingraham, 2018-08-21:
Learning a bit more about TCP/IP networking "nuances", the revelation to me has been that TCP protocol does seem to require a more "robust" connection. Thus Wi-Fi is connected... but the TCP packets are not being transported very well. So I started digging deeper.

I have a Google WiFi mesh. For whatever reason, some 802.11 b/g/n devices seem to want to connect to the primary WiFi Point rather than the closest point with the strongest signal. This isn't the case always, but it does seem to be the case usually. Therefore, my Sonoff devices in the back of my house are ending up connecting to the WiFi Point furthest from them... and thus a poor Wi-Fi connection resulting in dropped TCP connections.

So, I just reset my entire Google WiFi network (rather than just one WiFi Point and the respective Sonoff switches) and reset all the Sonoff devices in the back of the house... and they are now (currently) connected to the closest WiFi Point to them. RSSI values right now are "healthier". I'll monitor to see if the connection to the MQTT broker is more stable.