arendst/Tasmota #3534: Home assistant auto discovery not working

kukachik, 2018-08-20:
Hi,
Do anybody have a problem with hass discovery ?
In the last git version, and in 6.1.1 version I have this -
```
00:00:00 Project sonoff sensors-sonoff-7928 (Topic sensors-sonoff-7928, Fallback DVES_0DBEF8, GroupTopic sonoffs) Version 6.1.1.7-2_3_0
00:00:00 WIF: Connecting to AP1 kukachik in mode 11N as sensors-sonoff-7928...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sensors-sonoff-7928.local with IP address 192.168.0.68
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sensors-sonoff-7928/LWT = Online (retained)
00:00:05 MQT: cmnd/sensors-sonoff-7928/POWER = 
00:00:05 MQT: tele/sensors-sonoff-7928/INFO1 = {"Module":"Generic","Version":"6.1.1.7","FallbackTopic":"DVES_0DBEF8","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sensors-sonoff-7928/INFO2 = {"WebServerMode":"Admin","Hostname":"sensors-sonoff-7928","IPAddress":"192.168.0.68"}
00:00:05 MQT: tele/sensors-sonoff-7928/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_1/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_1/config =  (retained)
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_2/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_2/config =  (retained)
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_3/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_3/config =  (retained)
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_4/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_4/config =  (retained)
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_5/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_5/config =  (retained)
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_6/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_6/config =  (retained)
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_7/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_7/config =  (retained)
00:00:05 MQT: homeassistant/light/sensors-sonoff-7928_8/config =  (retained)
00:00:05 MQT: homeassistant/switch/sensors-sonoff-7928_8/config =  (retained)
```
If I setting "SetOption19 1" in console, I get 

```
homeassistant/light/sensors-sonoff-7928_1/config (null)
homeassistant/switch/sensors-sonoff-7928_1/config (null)
homeassistant/light/sensors-sonoff-7928_2/config (null)
homeassistant/switch/sensors-sonoff-7928_2/config (null)
homeassistant/light/sensors-sonoff-7928_3/config (null)
homeassistant/switch/sensors-sonoff-7928_3/config (null)
homeassistant/light/sensors-sonoff-7928_4/config (null)
homeassistant/switch/sensors-sonoff-7928_4/config (null)
homeassistant/light/sensors-sonoff-7928_5/config (null)
homeassistant/switch/sensors-sonoff-7928_5/config (null)
homeassistant/light/sensors-sonoff-7928_6/config (null)
homeassistant/switch/sensors-sonoff-7928_6/config (null)
homeassistant/light/sensors-sonoff-7928_7/config (null)
homeassistant/switch/sensors-sonoff-7928_7/config (null)
homeassistant/light/sensors-sonoff-7928_8/config (null)
homeassistant/switch/sensors-sonoff-7928_8/config (null)
stat/sensors-sonoff-7928/RESULT {"SetOption19":"ON"}
```
in mqtt.

My version-
```
Program Version | 6.1.1.7
Core/SDK Version | 2_3_0/1.5.3(aec24ac9)
```

Its my problem or problem in Tasmota ?

ryanbeaton, 2018-08-21:
My H801 auto discovery has done the same, showing up as a switch not a light :(

00:00:07 MQT: tele/sonoff/INFO1 = {"Module":"H801","Version":"6.1.1","FallbackTopic":"H801_4BD4EB","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5355","IPAddress":"192.168.201.154"}
00:00:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: homeassistant/light/sonoff_1/config =  (retained)
00:00:07 MQT: homeassistant/switch/sonoff_1/config = {"name":"H801","command_topic":"cmnd/sonoff/POWER","state_topic":"stat/sonoff/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoff/LWT","payload_available":"Online","payload_not_available":"Offline","brightness_command_topic":"cmnd/sonoff/Dimmer","brightness_state_topic":"stat/sonoff/RESULT","brightness_scale":100,"on_command_type":"brightness","brightness_value_template":"{{value_j ... (retained)

ryanbeaton, 2018-08-21:
My H801 auto discovery has done the same, showing up as a switch not a light :(

```
00:00:07 MQT: tele/sonoff/INFO1 = {"Module":"H801","Version":"6.1.1","FallbackTopic":"H801_4BD4EB","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5355","IPAddress":"192.168.201.154"}
00:00:07 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
00:00:07 MQT: homeassistant/light/sonoff_1/config =  (retained)
00:00:07 MQT: homeassistant/switch/sonoff_1/config = {"name":"H801","command_topic":"cmnd/sonoff/POWER","state_topic":"stat/sonoff/RESULT","value_template":"{{value_json.POWER}}","payload_off":"OFF","payload_on":"ON","availability_topic":"tele/sonoff/LWT","payload_available":"Online","payload_not_available":"Offline","brightness_command_topic":"cmnd/sonoff/Dimmer","brightness_state_topic":"stat/sonoff/RESULT","brightness_scale":100,"on_command_type":"brightness","brightness_value_template":"{{value_j ... (retained)
...
```
I think I'll go back to v5.14.0

arendst, 2018-08-21:
I love the people complaining before reading at least the Release information.

The answer, as most of the times, is in there AND STOP GOING BACK if something doesn't do what you expect. Think out of your little box and start to explore the world. It's wonderful!

ryanbeaton, 2018-08-21:
My apologies for offending you. I understand why that is happening now RE: SetOption30

arendst, 2018-08-21:
No problem. I just had to ventilate. Sorry. Glad you found it.

kukachik, 2018-08-21:
But what with sensor component and (retained) ?
I dont have any light/switch, only sensors.
Do discovery support sensor component at all?

...

kukachik, 2018-08-21:
Asking the right question is half the answer. I found answer in (#1534)
Only light/switch supported.

kukachik, 2018-08-23:
With H801 I got this error in hass
```
Traceback (most recent call last):
  File "uvloop/cbhandles.pyx", line 66, in uvloop.loop.Handle._run
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/mqtt/__init__.py", line 691, in _mqtt_handle_message
    msg.topic, payload, msg.qos)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/core.py", line 300, in async_run_job
    target(*args)
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/light/mqtt.py", line 273, in rgb_received
    self._hs = color_util.color_RGB_to_hs(*rgb)
TypeError: color_RGB_to_hs() takes 3 positional arguments but 5 were given
```
Because of 
```
"Color":"0,255,0,0,0"
```

And only light state could be see in hass. Not a color.