arendst/Tasmota #3535: Sonoff 4ch pro r2 with physical switch

Vrnreg, 2018-08-20:
Hi,

I have have few sonoff 4ch pro r2 with me.
I've seen some YouTube videos where sonoff can be used in sync with physical switches (when the physical switch is turned on, sonoff relay will also be turned on and vice versa). I heard that it is possible with tasmota.
I've checked few threads here and I was not able to find a solution. Do we need to solder the relay wire to some of the GPIO pins for this or just flashing with tasmota will do the trick?

Can anyone help me by providing an insight on how to do this?

Thanks a lot in advance.

Jason2866, 2018-08-20:
A 4 ch pro isnt a good choice for connecting additional switches.
The unit needs for his function nearly all Gpios that the ESP chip has.
Only Gpio 1,2 and 3 are left and this Gpios cant be used for connecting switches.

Frogmore42, 2018-08-20:
@vrnreg be aware that while people use Sonoff Basic devices attached to existing switches in their house, this is hard to do safely, and likely would not meet any code requirements.

Many people don't know or care about code requirements or safety.  If you do, I would suggest you use something like the T1.  While it is not UL listed, it is probably significantly safer than anything you are likely to wire up.

While I like the Sonoff devices, they are cheaply designed and manufactured.  This is fine when nothing unusual happens, and it also makes them cheap to buy.  But, if something unusual happens (and it likely will at some point) they will fail.  The failure might be realtively benign and only render the device non-functional.  Or, it might be more catastrophic and cause significant damage.  I would never hardwire one of them into my house.  But, I have several that I use that I plug in and can easily remove and replace.

There was someone who had two TH-16s fail with excessive heating when used to control a boiler that only draws 8A (half of the rating of the device).

Most lighting circuits are now pretty low power (assuming you are using LEDs), so fire is less of a concern for lighting circuits.

Vrnreg, 2018-08-21:
Yeah. I'm planning to use them only for lights and fans. I have a sonoff POW which I'll use for TV. I guess that shouldn't do anything any damage. Isn't?

ascillato, 2018-08-23:
> I have a sonoff POW which I'll use for TV. I guess that shouldn't do anything any damage. Isn't?

It is ok, if your sonoff POW is standard. No hardware modification. Sonoff POW is very dangerous is you add sensors or switches to it.

ascillato, 2018-08-23:
If your questions has been answered, please close this issue. Thanks

Vrnreg, 2018-08-24:
Thanks a lot for the support. My queries had been answered.

closed 2018-08-24 by Vrnreg