arendst/Tasmota #3536: Access admin mode offline - Captive Portal

tbagro, 2018-08-20:
I have a Sonoff Basic with tasmota 6.1.1, and I need to use it in stand-alone offline mode, but when I access Captive Portal (192.168.4.1 4x button config),  it tries to reconnect to a network and can not access the configuration of the card.

How do I modify the code to access the webserver in administration mode in AP mode through Captive Portal?

ascillato, 2018-08-21:
Hi,

> I need to use it in stand-alone offline mode

Sorry. That is not the objective of Sonoff-Tasmota. 

From the readme:

> Sonoff-Tasmota
> Alternative firmware for ESP8266 based devices like iTead Sonoff with web, timers, 'Over The Air' (OTA) firmware updates and sensors support, allowing control under Serial, HTTP, MQTT and KNX, so as to be used on Smart Home Systems.

It is intended to be used in Smart Home Systems.

closed 2018-08-21 by ascillato2

philfleck, 2023-02-10:
You might want to look here

https://github.com/arendst/Tasmota/discussions/13362#discussioncomment-4932672