arendst/Tasmota #3537: Sonoff basic with 3 reed sensors (magnetic contacts) for A/C shut-off

krasatos, 2018-08-20:
Hey, 
I have hooked up my sonoff on a power relay that controls an A/C unit.
I have also hooked up 3 magnetic contact sensors (set up as switch2,3,4) that i have placed on 3 windows in the same room as the A/C

My desired result is for the sonoff to power the a/c only when all 3 contacts are closed.

Is there some available command configuration to set this up without rules? (I have managed to set it up on a similar setup with a sonoff an one magnetic contact. Im having trouble configuring multiple contacts.)

I have noticed that there is no IF functionality on the rules section yet. Is there any way to achieve this?

Finally, I  have been trying to de-associate the sensors from the relay by using switchtopic, in order to send the messages over to a home assistant instance and do the job there but i had no luck unfortunately, whatever switchtopic i use, the sensors keep firing the relay.

Please give me some input, or direct me towards some reading material.

Thanks in advance!

ascillato, 2018-08-20:
> My desired result is for the sonoff to power the a/c only when all 3 contacts are closed.

Connect them physically in series.

krasatos, 2018-08-20:
Hey,
> Connect them physically in series to only one sonoff input
That's what im going to do, just redid my circuit and it works ok!

As for the switchtopic, i have them set as switches, but for some reason i cannot de-associate them.
> (set up as switch2,3,4)

Ill go ahead with the different circuit for now, but i'd like to have an answer concerning 3 different sensors.
Also i've been reading the resources you linked all night and i have prior experience deassociating (pir sensors for example) but this one here just denies to work :)

Anyways, thanks for the immediate reply.
And i think that some functionality to be able to use sonoffs as expansions on home alarm system would be really popular. :)
Thanks again!

closed 2018-08-20 by krasatos