arendst/Tasmota #3543: Info required : Im slightly confused at the output of the RFBridge

asantaga, 2018-08-21:
Hi all,

this is really a "am i understanding this right" query..

OK, Scenario, i have a number of kerui d026 Door/Window sensors around my house.. 

Currently I have 
1. open/Close the window
2. Look at the Serial output in the RFBridge Console
3. Look at the "Data" field, which appears to be a unique value from  each device I have , depicting Open/Close for that device
4. in Home Assistant I setup a MQTT Sensor, using the Data field to determine if the sensor is open closed.

So , in my case , I open/Close the window and I see the following messages

```
Line 1: 16:11:45 SER: Received A4 24 EA 01 36 03 B6 B1 E1 E7 55
Line 2: 16:11:45 MQT: tele/sonoffRFBridge/RESULT = {"RfReceived":{"Sync":9450,"Low":310,"High":950,"Data":"B1E1E7","RfKey":"None"}}
Line 3: 16:11:48 SER: Received A4 25 1C 01 4A 03 A2 B1 E1 EE 55
Line 4: 16:11:48 MQT: tele/sonoffRFBridge/RESULT = {"RfReceived":{"Sync":9500,"Low":330,"High":930,"Data":"B1E1EE","RfKey":"None"}}
```

From what I know, 
Line1 is the raw data transmission for my window open
Line2 is tasmota parsing this 
Line3 is the raw data transmission for my window close
Line4 is tasmota parsing this

I have used the "Data" field as the one to look for, and in my case for a single sensor Data=B1E1E7 = Window Open and B1E1EE=Window closed.

So my questions now are
Q: What are the sync,low,high fields for ?
Q: Apparently the sensor also sends a "battery low" message, would this be a different Data value?

Am i doing this right?

arendst, 2018-08-21:
Yes you're doing it right.

The Sync/Low/High are timings and just for information. It's being detected by the receiver chip and passed on. You may notice slight differences in these values due to receiving lock-in but the Data should stay the same.

asantaga, 2018-08-21:
Cool.. I suppose I need to find a dead battery and test out the sensors to get the dead battery data... Shame they're already on the doors :-( 

 Also from reading the docs, the RFBridge only detects  PT2260, PT2262, PT2264 and EV1527 Transmitters, these appears to be commonly used in various devices..  

Cool, I'll close this issue/info, unless someone has something to add

thanks arendst , awesome quick response

closed 2018-08-21 by asantaga