arendst/Tasmota #3545: iFan02 Unable to Pair Remote

jumblies, 2018-08-21:
**Describe the bug**
Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : iFan02 with 2.4 GHz remote 
- [x] Searched google groups
- [x] Provide the output of command ``STATUS0`` :
```
21:32:25 MQT: stat/iFan1/STATUS = {"Status":{"Module":44,"FriendlyName":["iFan1"],"Topic":"iFan1","ButtonTopic":"0","Power":6,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
```

**To Reproduce**
Difficult without dissociating remote from iFan02 Unit.  But, pairing functions on iFan02 do not work with Tasmota Firmware.  

**Expected behavior**
Long press on iFan02 Unit should cause remote module (daughterboard) to enter pairing mode.  Per user guide [itead link](http://ewelink.coolkit.cc/?p=1790)

**Additional context**
To PAIR remote with iFan02 Unit requires use of button, but the button is coupled with Tasmota functions which appears to be interfering with re-pairing the remote with the iFan02 Unit.  There is no way to disable the button in Tasmota.  It may be that Tasmota FW is intercepting button press for tasmota behavior and remote module not being triggered for pairing.  This may be a problem with the daughterboard referenced in #3395.

Maybe a disable switch state could be used?  

The other option is removal, flashing of factory FW, pair remote, restore Tasmota.  :-(

arendst, 2018-08-23:
As the button is configured to GPIO0 you might want to try to temporary change the ifan module configuration from iFan02 to Generic. This way the button is not handled by Tasmota and you must be able to pair the remote.

I haven't tried it though ;-)

jumblies, 2018-08-23:
Great thought.  I tried it and no dice.  Can't find my factory firmware either.  Poo.

arendst, 2018-08-24:
I'm out of options.

I've tried some other things like using the led on GPIO13 to tell the microcontroller wifi is present (see schematics) but both settings do not trigger pairing either.

Anyone with a solution other than using the original firmware is welcome.

NB As the pairing is handled by the microcontroller I see no reason why the original firmware (on the esp8266) would solve the issue.

jumblies, 2018-08-24:
No worries.  I'm going to ask and see if has the factory firmware  and give it a whirl.

closed 2018-08-24 by jumblies

jumblies, 2018-08-24:
@arendst 

Thank you for your patience.  It was an ID 10 T error.  Apparently if the voltage in the button cell hits 2.9, that's enough to cause it to malfunction.  

UPDATE: HOLY CRAP, the oldest trick in the book.

The battery cell went bad. It was 2.9 V on the voltmeter which I figured would be fine. Even tried a couple of different ones. I can't believe I fell for that. FACEPALM

janneslr, 2018-09-27:
Hi, I have exactly the same issue, but my battery measures 3.1V.  Any suggestion

janneslr, 2018-10-05:
Hi, just an update on this.  I manage to sort out my issue.  The device needs to be connected to AC power for the remote do function.

jumblies, 2018-12-27:
> any other ideas to try here? Remote stopped working when i flashed with
> sonata.exe
> I am powering it from 240v cell battery is at 3.2v i can't seem to get
> anything from it, the several different repairing instructions from power
> cycling the sonoff while holding the pair button and battery in place, to
> holding the button on the sonoff, nothing seems to help.

I found the remote to be very fickle with batteries and got both of mine
working again.  I think the remote have a quiescent power draw that causes
problems.  On one, I put a switch to cut battery when I'm not using it.

My advice based on my experience is to try several different batteries and
the standard pairing procedure.  It's highly unlikely to be an issue with
tasmota or the ESP. The battery caused such confusing behavior, I became
convinced it was something else.

Wihanh, 2018-12-27:
> I found the remote to be very fickle with batteries and got both of mine working again. I think the remote have a quiescent power draw that causes problems. On one, I put a switch to cut battery [...]

Thx, can you clarify the exact procedure you used to repair? and can i confirm you did it with Tasmota loaded?

jumblies, 2018-12-27:
I did it with Tasmota.  I even flashed the original firmware but it didn't change anything.  As others have noted, the remote is completely separate subsystem that just feeds info to the ESP.  

I left my Tasmota configured functionally and confirmed all activity over MQTT before fiddling with anything else.  I recommend that so it removes a variable. 

Reflowed the daughter board pins as others suggested.  No difference.

Replaced the battery holder with one from Ali in case that was the issue.  No difference

Replaced the battery with a brand new battery and pressed the button on the front of the remote until the fan unit beeped.  That's it.  Just it is supposed to work.  I was surprised and felt stupid but these remotes are really flaky.  Erratic behavior if the cell isn't full.  

 think I measured mA draw at one point but can't remember what it was.  I think that if there is any undue internal resistance in the battery, the remote misbehaves.

jumblies, 2018-12-28:
> So you didnt power cycle or press anything on the fan controller only on
> the remote?
>
> Which button did you press the hidden silver disc inside or just one of
> the remote buttons?

I tried the power cycle holding the hidden button on the back of the opened
remote but there wasn't any indication that it did anything.

I used the standard pairing method power instructions using the front
button.

I feel your pain.  It's frustrating.

slavekjurkowski, 2018-12-30:
Can confirm, experienced same pairing issue as described. I replaced the battery with a fresh cr2032 which is .7mm thicker. slightly more battery capacity and paired right away.

jumblies, 2018-12-30:
bottom button front of remote.  In another longer ifan thread I posted a picture of the instructions.

Wihanh, 2019-01-21:
Yeah, not sure if I am just a idiot or my remote or unit is faulty, but i can't get anything out of it.
I tried with the thicker CR2032 battery.
I tried pushing the hidden disk as lanefu mentioned, i tried pressing it and power cycling, just pressing it, pressing it and then pressing other buttons while it is flashing.
I tried pushing the bottom button on remote while unit is powered, i tried holding it while power cycling.
I tried just pressing it a bunch of time with the unit powered on.
The unit is powered with 240v from the mains.  It works fine from mqtt, but nothing from the remote.
I haven't reflowed the daughter board, but i think i might just order another unit and see it is just this one that is having a problem.
BTW, the guy in the video mentioned above seems like it is not working, because you can't here the relays click afterward, so i am guessed this is a video he made to show itead how it doesn't work.
I commented on it anyway 3 weeks ago asking him if it worked, but there was no response.

Wihanh, 2019-01-21:
i see itead website now says retired for this unit.

iJuiceman, 2019-02-15:
I also had problems with the remote not working, but the new 2032 battery fixed the problem.

fquinto, 2019-03-16:
After a long time trying for me the procedure has been:
- Using the original firmware (from Itead Sonoff).
- Check battery: not less 2.85v and not 3.3v. Only runs very near 3v.
- Put the new battery: press the hide button (open remote controller for that)
- The LED is like "searching"
- Connect the ifan02 to the AC current or if connected, reboot.

I think the join during the reboot of the main unit.

Hope this helps.

HadiHassaoui, 2019-04-04:
I have two iFan02 modules, one in living room, and one in Bedroom.

The one in the living room was installed first, and was paired with the remote for one week until I installed the Bedroom fan. I manage to pair the new remote with the Bedroom fan. 

But then the LivingRoom remote won't connect to the fan. Based ont fquinto's solution I did the following steps:

* Using Tasmota 6.5.0
* Change battery for CR2032
* Turn of the power (breaker)
* Press the Hidden button (made a hole in the plastic to use it with a pin)
* Turn on the power (breaker)
* The flashing stopped immediately
* Press the Pair button (the one the kind of WIFI icon)
* Pairing successfull.

Then, Surprise the remote in the Bedroom won't pair anymore. Did the exact same procedure for the bedroom (without changing battery), and BINGO both remote are paired.

For me the problem came from having two iFan02 module.

Sineos, 2019-04-21:
For future reference: Restarting Tasmota did not work for pairing. Mine needed a complete power cycle:

1. AC mains off
2. Press hidden button on the remote
3. AC main on