arendst/Tasmota #3546: WeMos Button

svh1985, 2018-08-21:
Hello,

I was just wondering.. how can I use the button features as described on (https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage) with the WeMos D1 mini? I've connected the default button shield (pin D3) but it does not work. Do I need to do some additional config?

Thanks!

Frogmore42, 2018-08-22:
Yes, you need to configure the GPIO to be a button.  See here for section of the UI that does it.
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini#generic-module

ascillato, 2018-08-22:
Hi, 

If you think that your question has been addressed, please close this issue. Thanks

svh1985, 2018-08-22:
Hey, I've setup the module as followed.. but the button press doesn't trigger anything;
[image]

ascillato2, 2018-08-22:
You did not set any output. Please, read the wiki.

As a basic example use:

Button1 in gpio0
led1i in gpio2
Relay1 in gpio12

svh1985, 2018-08-24:
Two more questions; 
1. I only want to use the button to control the 1 short press, 2 short press etc. actions. Do I need to set the relais(output) in order to use the button functions?
2. Is it possible to code these settings in the firmware as default? I don't want to specify wifi in the firmware but rather use the button action to launch the WiFiManager and configure wifi.

Thanks!

closed 2018-08-24 by ascillato2