arendst/Tasmota #3547: RGB support added to domoticz but not working with tasmota

cobradevil, 2018-08-22:
I have seen different comments and commits on adding support for RGB dimmer within tasmota but there was no support within domoticz. With the latest release of domoticz this has changed but I still can't get it to work.

When i add an rgb dimmer within domoticz and set the color i get the following MQTT output:
{
   "Battery" : 255,
   "Color" : {
      "b" : 88,
      "cw" : 0,
      "g" : 50,
      "m" : 3,
      "r" : 255,
      "t" : 0,
      "ww" : 0
   },
   "Level" : 56,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Color Switch",
   "id" : "00082489",
   "idx" : 489,
   "name" : "rgb-tasmota",
   "nvalue" : 10,
   "stype" : "RGB",
   "svalue1" : "56",
   "switchType" : "Dimmer",
   "unit" : 1
}

I have flashed the git version of tasmota and configured it as a magichome device and provided  setoption15 1 on the console. I configured the mqtt server. The basic on of command is received but it seems that the color command is ignored.

power on command produces the following output:
{
   "Battery" : 255,
   "Color" : {
      "b" : 19,
      "cw" : 0,
      "g" : 247,
      "m" : 3,
      "r" : 255,
      "t" : 0,
      "ww" : 0
   },
   "Level" : 47,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Color Switch",
   "id" : "00082489",
   "idx" : 489,
   "name" : "rgb-tasmota",
   "nvalue" : 1,
   "stype" : "RGB",
   "svalue1" : "47",
   "switchType" : "Dimmer",
   "unit" : 1
}

Any ideas or thoughts on this one?

Best regards,
William van de Velde

arendst, 2018-08-22:
Well as you said Domoticz is just supporting the color command chances are that Tasmota does not understand Domoticz,

In fact up till now the Domoticz color data is gibberish for Tasmota.

Might be implemented in the near future.

cobradevil, 2018-08-22:
@arendst 
I did a small donation for your work.

Best regards
William

arendst, 2018-08-22:
Don't push me ;-)

What version of domoticz does support this color command. Is it the latest stable one or a newer beta one?

cobradevil, 2018-08-22:
haha, I won't
This is the latest stable version of domoticz.

arendst, 2018-08-22:
OK. Will take some time as I have to upgrade Debian first before I can run the latest stable. All I need is time, time, time.

cobradevil, 2018-08-22:
No problem, got all day ;-)

cobradevil, 2018-08-22:
Hello Theo,

so it seems that when using the dummy device within domoticz to add the dimmer manually it does not work except for the on of command but when using the mqtt discovery plugin from @emontnemery  it does seem to work. I do not have the right colors right now but will change this tonight to see if it will work.

Regards,
William

emontnemery, 2018-08-22:
I can have a look at this if you don't have time @arendst, I have everything setup already.

arendst, 2018-08-22:
@emontnemery please do as I'm still busy with the display stuff.

emontnemery, 2018-08-22:
@arendst What's you view on adding more features to Tasmota's Domoticz extension which relies on creating Dummy devices in Domoticz by the way?

The reason I wrote the plugin in the first place is that the Dummy device method:
- It very laborious when adding new devices, requiring multiple setup steps in both Domoticz and Tasmota
- Is quite unreliable: If the Tasmota device is down, this is not reflected in Domoticz UI

@cobradevil Do you mean that after the correction of the plugin to support "lights" identifying as a HASS "switch", you can suddenly control the color through the manually created dummy device _as well as_ through the device created by the plugin?

cobradevil, 2018-08-22:
@emontnemery What i mean is that when adding the light to domoticz using the dummy option and manually select rgb dimmer and put the idx in the tasmota config then i can only set the light on and off within domoticz and no colors are changed. But when using the mqtt discovery added dimmer, then I can select colors (not all off them because red is not working yet, but it does change colors) which is not the case when using the manually added option.

emontnemery, 2018-08-22:
Ok, that makes sense:
The plugin supports translating Domoticz color command to corresponding Tasmota commands
Tasmota's Domoticz extension doesn't have this support

ascillato, 2018-08-23:
Just for Reference: Similar issue https://github.com/arendst/Sonoff-Tasmota/issues/734

closed 2018-08-24 by arendst

cobradevil, 2018-08-26:
The problem looks solved except that red and green colors are swapped when controlled from domoticz.
When i pick the red color in domoticz then it turns green and the other way around.

When i manually connect the wires from the rgb strip i can see the mapping is right.

emontnemery, 2018-08-26:
@cobradevil Do you get wrong color both when using the plugin and when using the manually added dummy device?

cobradevil, 2018-08-26:
yes that is the same. so only the red and green are reversed

emontnemery, 2018-08-26:
OK, then the problem is your HW...
Most likely PWM channel assignment in Tasmota is not matching your HW.

cobradevil, 2018-08-27:
ok so I have changed the pwm assignment within the magichome module and it works ok.

Would there be a possibility to select the channel assignment within the GUI instead off manually compiling the firmware with the assignment configured statically?

Thank you so far.

Best regards,
William van de Velde

arendst, 2018-08-27:
Regarding MagicHome/Arilux hardware it seems that every device has a different PWM/GPIO relation which makes pre-configured choices impossible. Tasmota does support a few but apparently yours is not one of them.

You could select the Generic module and add your known PWM channels to the correct GPIOs. Tasmota will configure itself according to the amount of PWM channels configured. This way you do not have to change hardware and can still use Tasmota as desired.

Hope that makes sense.

cobradevil, 2018-08-27:
That makes sense indeed but can it be made possible to turn on and off a gpio without saving / rebooting all the time?
So when i select the generic module and set gpio12 on i can see the light turning on immediately without having to reboot all the time? Otherwise it would be a horrible task to setup if you do not know the pin layout because you have eleven options. And is pwm1 always red pwm2 green and pwm3 blue and pwm4 white? maybe those names could then be added to pwm1 so it would be much easier to configure the generic module with a magichome like module.

I hope that makes sense ;)

arendst, 2018-08-27:
Would be nice but any changes to the GPIO config influence the automation within Tasmota so a reboot is needed to get things in place.

See file sonoff_template.h:
```
  GPIO_PWM1,           // RGB   Red   or C  Cold White
  GPIO_PWM2,           // RGB   Green or CW Warm White
  GPIO_PWM3,           // RGB   Blue
  GPIO_PWM4,           // RGBW  (Cold) White
  GPIO_PWM5,           // RGBCW Warm White
```
PWM1 is Red when PWM3 is also configured otherwise it's Cold White as there are only two PWMs
PWM2 is Green when PWM3 is also configured otherwise it's Warm White as there are only two PWMs

cobradevil, 2018-08-27:
I see what you mean but there is no other way to enable gpio's temporarely e.g. from the console?

arendst, 2018-08-27:
You can use command ``Gpio`` but it has the same obvious result; once executed it will restart the device to reconfigure.

PR #3569 (2018-08-24): Add support for setting color from Domoticz
Closes #3547 and closes #734 
All modes supported by Domoticz are working (WW, RGB, RGBW, RGBWW)