arendst/Tasmota #3548: Half hour time zone Australia CST

maxduckduck, 2018-08-22:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : sonoffusers pow2
- [ x] Provide the output of command ``status 0`` :
```
{"Status":{"Module":43,"FriendlyName":["Sonoff001"],"Topic":"sonoff001","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
```
Hi
I can set time zone to full hour increments, but I would like a half hour increment (Australian CST +9:30).  How is this done?

curzon01, 2018-08-22:
Set **TimeSTD** and **TimeDST** as desired (it allows offset from UTC in minutes) and use `Timezone 99` - see [Management Commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management)

ascillato2, 2018-08-22:
Duplicated issue
#2684

closed 2018-08-22 by ascillato2

maxduckduck, 2018-08-22:
I do not want Dst, do I just set start and stop days to the same?

ascillato, 2018-08-23:
Yes