arendst/Tasmota #3550: CVE-2018-6692 - Potential Remote Execution and arbitrary access to remote LAN.

Aterfax, 2018-08-22:
**[CVE-2018-6692 ](https://nvd.nist.gov/vuln/detail/CVE-2018-6692)**

Further detail available [here](https://securingtomorrow.mcafee.com/mcafee-labs/insight-into-home-automation-reveals-vulnerability-in-simple-iot-product).

A flaw in Belkin Wemo smart switches has been found which allows remote execution (Stack-based Buffer Overflow) and the ability to compromise the smart switch to allow arbitrary access to users's LANs remotely.

The vulnerability resides in libUPnPHndlr.so, is exploited via the device requesting an open port via UPNP and subsequently being attacked via a crafted HTTP post packet.

Belkin Wemo emulation of the Tasmota firmware may be vulnerable depending on the implementation of Wemo switch emulation. 

I don't have time to investigate whether Tasmota's emulation of the Belkin Wemo switch is vulnerable (or even uses UPNP) but I thought you would want to know.

- [X] Searched the problem in issues and in the wiki
- [N/A] Hardware used : 
- [N/A] Provide the output of command ``status 0`` :

**To Reproduce**
Not known if vulnerable.

**Expected behavior**
If vulnerable, remote access possible to the Wemo switch and the LAN in which it resides.

**Screenshots**
N/A

**Additional context**
N/A

ascillato, 2018-08-22:
Hi,

Thank you very much for sharing this.

I have take a look at the report and the exploit uses a vulnerability of the internal libraries of wemos. Tasmota don't use those. So that exploit should not work IMHO. We can take a deep look into it.

Thanks again.

Aterfax, 2018-08-22:
Bueno!  :ok_hand:  Good to know.

closed 2018-08-22 by Aterfax