arendst/Tasmota #3553: Sonoff RF - no WIFI after RF programming

Friday13th87, 2018-08-22:
Hey there!
i am using a Sonoff RF with the newest tasmota firmware.
after programming the RF controller and replugging the sonoff to a socket, it is not connecting to wifi anymore (before it was working of course - and i controlled it via openhab)
Its only "klack" and blinking anymore. RF is working.

in the wiki for the RF is written:
-2 times short is programming
-3 times short is forgetting RF

on the normal wiki is written that pressing the button 2 times means something like "special function" and 3 times putting the device in wifi config mode.

i dont get it why where is not more information or does this mean that for forgetting RF i need to press the button two times and afterwards another one time to come out of wifi config mode?
or does the normal button stuff doesnt count for the RF unit?

after reflashing the unit its always working again, till i program the RF controller.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Friday13th87, 2018-08-23:
Short Update - it was "klicking" till i went to bed. today in the morning the device was connected sucessfully and is showing up in openhab and the webpage is working as well.

i dont know what was happening - as the sonoff is only ~ 1.5m away from the router it was shouldt be a wifi signal problem.

ascillato, 2018-08-23:
So,

Your issue has been solved?

ascillato2, 2018-08-27:
Hi,

Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-08-27 by ascillato2