arendst/Tasmota #3554: Option to sleep only when lights are off

gitolicious, 2018-08-22:
**Have you looked for this feature in other issues and in the wiki?**
_yes_

**Is your feature request related to a problem? Please describe.**
_When enabling sleep on my [MagicHome LED strip](../wiki/MagicHome-LED-strip-controller), I get flickering lights every now and then. This is known and described in [Energy Saving](../wiki/Energy-Saving)._

**Describe the solution you'd like**
_If sleep would only be performed when my lights are off, it would not affect the LEDs._

**Describe alternatives you've considered**
_Completely disabling sleep misses the point of reducing power usage._

**Additional context**
_-_

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-08-22:
Hi,

This can be achieved using rules.

Please, read RULES in the wiki. Thanks.

gitolicious, 2018-08-22:
Are you sure that would work? The general idea sounds reasonable, but when sending the `sleep` command manually, my device reboots. If I now say
`Rule1 on Power1#State=1 do sleep 0 endon`
`Rule2 on Power1#State=0 do sleep 200 endon`
I guess this would lead to a reboot every time I switch the lamp on or off...
(Sorry if I got the syntax wrong, never tried rules before)

ascillato, 2018-08-22:
Hi, I tested

```
rule1 1
rule1 on Power1#State=1 do sleep 0 endon on Power1#State=0 do sleep 1 endon
```

and works without problem.

gitolicious, 2018-08-23:
Thanks for trying this out! I will try as soon as I am back home.
Some issues I expect by this solution:
* Weird dimming/wakeup behavior when having enabled `fade` and `speed` options as during the reboot fading will stop
* Local Tasmota logs are lost after reboot
* Remote Tasmota logs are flooded with reboot messages
* Router logs are flooded with WiFi diaconnect/connect messages
* Debugging of this will be tricky due to the reboots
* Debugging of other stuff will be tricky due to the reboots
* Monitoring number of (unexpected) reboots usually is a good indicator if your system is running stable - won't be possible anymore 

Still I appreciate your idea and will give it a go next week.

ascillato, 2018-08-23:
@arendst 

Theo, why Tasmota have to reboot when changing sleep command?

In _sonoff.ino - line 563_ I change this original code:

```
    else if (CMND_SLEEP == command_code) {
      if ((payload >= 0) && (payload < 251)) {
        if ((!Settings.sleep && payload) || (Settings.sleep && !payload)) restart_flag = 2;
        Settings.sleep = payload;
        sleep = payload;
      }
      snprintf_P(mqtt_data, sizeof(mqtt_data), S_JSON_COMMAND_NVALUE_UNIT_NVALUE_UNIT, command, sleep, (Settings.flag.value_units) ? " " D_UNIT_MILLISECOND : "", Settings.sleep, (Settings.flag.value_units) ? " " D_UNIT_MILLISECOND : "");
    }
```

to this:

```
    else if (CMND_SLEEP == command_code) {
      if ((payload >= 0) && (payload < 251)) {
        Settings.sleep = payload;
        sleep = payload;
        if (Settings.sleep) { WiFi.setSleepMode(WIFI_LIGHT_SLEEP); }
      }
      snprintf_P(mqtt_data, sizeof(mqtt_data), S_JSON_COMMAND_NVALUE_UNIT_NVALUE_UNIT, command, sleep, (Settings.flag.value_units) ? " " D_UNIT_MILLISECOND : "", Settings.sleep, (Settings.flag.value_units) ? " " D_UNIT_MILLISECOND : "");
    }
```

and with debug on, seems that sleep is working.

It is any other thing needed that requires the reboot?

arendst, 2018-08-24:
The problem is in support.ino line 1099:
```
  if (Settings.sleep) { WiFi.setSleepMode(WIFI_LIGHT_SLEEP); }  // Allow light sleep during idle times
```
Currently if sleep is 0 the reboot will make sure that setSleepMode is not executed.

I will have to investigate if Wifi sleepmodes can be changed on-the-fly like disabling sleep.

arendst, 2018-08-24:
Give it a go. How on earth do you know if the device is sleeping?

ascillato, 2018-08-24:
I was just looking to the free cycles of the main loop with debug on. But I did not check the power consumption. That's why I said seems that sleep is working. I will try to measure power consumption.

arendst, 2018-08-24:
Are you able to run your device with core 2.4.2?

Mine hangs within a minute when I set sleep to 10. After a reboot the same happens so in that case the functionality is equal to before the last change.

Using core 2.3.0 the program keeps running fine...

Is there still a problem with sleep in core 2.4.2?

ascillato, 2018-08-24:
Ups!

Sleep have issues in v2.4.2. What a disappointment!

I will make a little sketch to test only sleep to confirm this, and then we can make a new issue at arduino repo. 

I was testing with sleep 1 and after leaving my testing devices, after 28 minutes they got stuck. No core dump by serial nor reset.

Then I test:
*  sleep 1 (about 28 minutes to stuck)
*  sleep 10 (about  2 minutes to stuck)
*  sleep 200 (about 2 seconds to stuck)

Another problem with sleep (but that is just Tasmota related and fixable) is that the LWT for MQTT is time stretched so, with sleep 1, you see MQTT disconnections and connections.

ascillato, 2018-08-24:
@gitolicious 

Coming back to the original topic of this issue.

In the actual development version, sleep command will not restart your device. It works great with esplib 2.3.0.

so using the rules:

```
rule1 1
rule1 on Power1#State=1 do sleep 0 endon on Power1#State=0 do sleep 1 endon
```

will gave you your desired behaviour.

digiblur, 2018-08-24:
Can can confirm as well that sleep is not working with 2.4.2.  Might be a good idea to put compile time code that makes the device not sleep if it is compiled with 2.4.2.  Save some support issues.

closed 2018-08-24 by ascillato2

ascillato, 2018-08-24:
I have tested (with last dev version) the following rule as a workaround for this sleep esplib v2.4.2 problem:
```
rule on system#boot do backlog sleep 10; ruletimer1 60 endon on Rules#Timer=1 do backlog sleep 0; ruletimer1 60; sleep 10 endon
```
and the device works without problems! :smile: 
(at least for the time I have tested, I will leave the device more time to see its behaviour)

Also, I test:
```
rule on system#boot do backlog sleep 1; ruletimer1 900 endon on Rules#Timer=1 do backlog sleep 0; ruletimer1 900; sleep 1 endon
```
And the device works.
(at least for the time I have tested, I will leave the device more time to see its behaviour)

So, this gave me the idea for a workaround in the code when using sleep in v2.4.2. Before the "stuck time", we can turn sleep to 0 and then go back to the previous value.

So, @arendst , what do you think ?

arendst, 2018-08-24:
I wait for your test results....

ascillato, 2018-08-24:
Finally, after testing in several devices with different parameters like:

```
rule on system#boot do backlog sleep 10; ruletimer1 60 endon on Rules#Timer=1 do backlog sleep 0; delay 20; ruletimer1 60; sleep 10 endon
```

I found that close to an hour, the device **_freeze_** and the Soft WDT don't trigger.

So,

Sadly I have to say that **_Sleep don't work on any esplib different than v2.3.0_** 

We have to disable sleep feature for any other library different than v2.3.0 :disappointed: and wait for next esplib release.

Jason2866, 2018-08-25:
I have long time test with sleep 1 with 2.4.2 (Tasmota 6.1.1.7) and no problem since 6 days on a sonoff basic with a connected MHZ19 that works too without a problem.

Jason2866, 2018-08-26:
@arendst and @ascillato 
I have uploaded 6.1.1.10 (with reenabling sleep for 2.4.2) and do the ascillato rule
`rule on system#boot do backlog sleep 1; ruletimer1 900 endon on Rules#Timer=1 do backlog sleep 0; ruletimer1 900; sleep 1 endon`
and no freeze.
[image]

```
12:49:55 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:49:55 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:49:57 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:49:58 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:50:58 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:50:58 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:50:59 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:51:01 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:51:01 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:52:01 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:52:01 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:52:02 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:52:04 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:52:04 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:52:17 MQT: tele/sonoff-71C254/STATE = {"Time":"2018-08-26T12:52:17","Uptime":"0T13:10:36","Vcc":3.442,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
12:52:17 MQT: tele/sonoff-71C254/SENSOR = {"Time":"2018-08-26T12:52:17","MHZ19B":{"CarbonDioxide":765,"Temperature":25.0},"TempUnit":"C"}
12:53:04 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:53:05 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:53:05 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:53:07 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:53:07 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:54:08 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:54:08 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:54:08 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:54:10 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:54:10 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:55:11 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:55:11 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:55:11 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:55:13 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:55:13 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:56:14 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:56:14 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:56:14 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:56:16 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:56:16 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:57:17 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:57:17 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:57:17 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:57:18 MQT: tele/sonoff-71C254/STATE = {"Time":"2018-08-26T12:57:18","Uptime":"0T13:15:37","Vcc":3.442,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:2A:BD:19"}}
12:57:18 MQT: tele/sonoff-71C254/SENSOR = {"Time":"2018-08-26T12:57:18","MHZ19B":{"CarbonDioxide":785,"Temperature":25.0},"TempUnit":"C"}
12:57:19 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:57:19 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:58:20 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:58:20 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:58:20 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:58:22 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:58:22 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
12:59:23 RUL: RULES#TIMER=1 performs "backlog sleep 0; delay 20; ruletimer1 60; sleep 1"
12:59:23 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"0 (0)"}
12:59:23 MQT: stat/sonoff-71C254/RESULT = {"Delay":20}
12:59:25 MQT: stat/sonoff-71C254/RESULT = {"T1":60,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
12:59:26 MQT: stat/sonoff-71C254/RESULT = {"Sleep":"1 (1)"}
```

arendst, 2018-08-26:
In my case sleep 200 on core 2.4.2 fails within a minute while it runs as expected on core 2.3.0.

Considering the discussions on the core site I keep sleep disabled for the wiki modem when core 2.4.1 and 2.4.2 is used.

Jason2866, 2018-08-26:
Yes, i think this good idea, if i go above 10 devices fails too...

gitolicious, 2018-08-29:
Wow, coming back from the weekend I didn't expect that much action. Thanks so much for looking into it. Interesting insights you got here, even though it got way more complicated than I thought.
Let's hope the next core version will fix the sleep issue as it looks like we are not alone on this.