arendst/Tasmota #3555: Support 7 segment display

qingz2004, 2018-08-22:
**Have you look for this feature in other issues and in the wiki?**
Yes.
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
Able to display temperature and humidity read from a sensor with a cheap 7 segment display.
**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._
https://www.letscontrolit.com/wiki/index.php/7_segment_display
**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2018-08-23:
why not use OLED?  (is already supported!)
128x64 Pixels from 2-5 Euros

or character LCDS (also supported)

qingz2004, 2018-08-23:
@gemu2015 Thanks! But I only need to display digits and 7 segment display is still much cheaper (less than half price).

Frogmore42, 2018-08-24:
There are oleds that are very small and currently supported.  It has taken a long time to get that support, so don't know when 7 segment will come.  But, will come sooner if you do the work to integrate it 😀.

qingz2004, 2018-08-24:
@Frogmore42 Got a link of a cheap OLED which is smaller than 7 segment and supported by Tasmota?
Thanks!

Frogmore42, 2018-08-24:
I haven't tried these yet, but I expect them to work with the new support.
https://www.amazon.com/MakerFocus-Display-SSD1306-3-3V-5V-Arduino/dp/B0761LV1SD

They are available cheaper directly from China, but I like 2-day shipping and have had good luck with Maker Focus products.  I also hate wasting time on stuff that doesn't work, so my definition of cheap includes how much time I need to spend getting it to work..

Frogmore42, 2018-08-24:
There are of course some very small 7-segment displays, but you didn't provide a reference for your definition of small.  I also like to be able to read my displays without needing to be closer than a few inches away.

qingz2004, 2018-08-24:
@Frogmore42 Thanks! I will try that. Hope it works out of box.
It is cheaper from Aliexpress, but I have to wait for around 2 months.

Jason2866, 2018-08-24:
@qingz2004 
I have ordered this some month ago and it is working with Tasmota.
https://www.aliexpress.com/item/1pcs-0-96-white-0-96-inch-OLED-module-New-128X64-OLED-LCD-LED-Display-Module/32640136921.html?spm=a2g0s.13010208.99999999.259.39c13c00rkUzgu

qingz2004, 2018-08-24:
@Jason2866 How clear is it compare to 7 segment LED when it displays digits?
Thanks!

Frogmore42, 2018-08-24:
The one I listed is similar to the one that @Jason2866  has used, only smaller (128x32 instead of 128x64).  Both displays are incredibly crisp, but they are quite small.  So depending on your age an eyesight, you might need glasses to read them :-)

gemu2015, 2018-08-24:
if you display only a few digits with increased font size it is perfectly readable

but there are also versions with 1.3 inch which are better readable at small fonts

qingz2004, 2018-08-24:
Perfect.
Thanks!

Jason2866, 2018-08-24:
You can use a LCD Display too. Typ 1602  or 2004 (3.3 Volt) and a usual I2C board.
LCDs are cheap too...

qingz2004, 2018-08-24:
@Jason2866 Are the LCDs supported?

Jason2866, 2018-08-24:
Yes👍

qingz2004, 2018-08-24:
Great! Thanks!

andrethomas, 2018-08-27:
@qingz2004 Hi, yes the IC2 LCD's work very well - I have a few of them in use since the implementation of the display code a week or so ago.

If you feel your issue has been addressed please close this issue :) Thx

closed 2018-08-27 by qingz2004