arendst/Tasmota #3557: Sonoff Basic doent toggle in webser mode

kaefer2, 2018-08-23:
Sonoff Basic doesn't toggle in webserver mode with IE11

When i click TOGGLE in the Sonoff webserver the diplay changes but the Sonoff doesn't switch.
After a few seconds thje display changes back. This is independent whether the Sonoff is ON or OFF.

The browser command http://192.168.0.12/cm?cmnd=Power toggle works well.

This bug does not appear in Firefox.

- A bug in the Java script of the webserver?
- A bug in IE 11?
- wrong/bad configuration of IE? 

Another problem:
. the Sonoff is ON
- when starting the webserver the display is OFF, that means the display doesn't reflect the actual status.

Windows 10 Pro Version 1803

Please help, I need it in an application which uses the webbrowser active-x in VB6.

ascillato, 2018-08-23:
check for javascript to be the latest version in your browser

kaefer2, 2018-08-23:
Javascript is enabled. I don't know whether it is possible to have different versions in IE11.
The problem is the function la in the script:
```
function la(p){
	var a='';
	if(la.arguments.length==1){a=p;clearTimeout(lt);}
	if(x!=null){x.abort();}
	x=new XMLHttpRequest();
	x.onreadystatechange=function(){
		if(x.readyState==4&&x.status==200){
			var s=x.responseText.replace(/{t}/g,"<table style='width:100%'>")
				.replace(/{s}/g,"<tr><th>")
				.replace(/{m}/g,"</th><td>")
				.replace(/{e}/g,"</td></tr>")
				.replace(/{c}/g,"%'><div style='text-align:center;font-weight:");
			eb('l1').innerHTML=s;}
		};
	x.open('GET','ay'+a,true);
	lt=setTimeout(la,2345);
	x.send();
}
```
This function is called on laoding the page and on clicking TOGGLE
if the webserver starts the page the display is always OFF even if the Sonoff is ON.
This is not the case with other browsers.

Maybe a bug or incompatibility in IE11 with the XMLHttpRequest?

I found this in [(https://caniuse.com/#feat=xhr2)]

_•In Internet Explorer the timeout property may only be set after calling open and before calling send._
>>> I have changed it, but no effect (see above)

_•IE10 and 11 do not support synchronous requests, with support being phased out in other browsers too. This is due to detrimental effects resulting from making them._

I'm not so familiar with Java so I need help.

closed 2018-08-23 by kaefer2

arendst, 2018-08-23:
IE11 is end of life for a long time. Consider other browsers like Edge, Chrome or Firefox.